sarnau/LoxLink #1: Entwicklungsumgebung

challo2018, 2019-07-07:
Hallo Markus,
 vielen Dank für dein Super Projekt. 
Ich kenne mich mit den ST32 Entwicklungsumgebungen bisher nicht sehr gut aus. 
Dein Projekt ist in Crossworks erstellt !? Ist es hierfür notwendig eine Lizenz zu kaufen oder gibt es auch kostenfreie Varianten?

Vielen Dank!
Grüße Fabian

sarnau, 2019-07-07:
Ja, das kostet Geld (150 Euro für non-commercial). SEGGER bietet quasi die gleiche SDK für Hobbyisten umsonst an, man muß lediglich die Extension vom Projektfile ändern.

challo2018, 2019-07-07:
Danke für den Tipp, mit Segger EmbeddedStudio funktioniert es auch. Allerdings gibt es Probleme bei der CTL Lib. Muss ich diese dann aus Crossworks includieren? Hast du hier Erfahrung?
Vielen Dank!

sarnau, 2019-07-07:
Ja, die kommt von Crossworks. Ein Grund warum ich Crossworks genommen habe, ist, weil Loxone auch in Crossworks entwickelt. Die CTL ist identisch.

challo2018, 2019-07-07:
Ich schaue mir erstmal an wie ich klar komme, dann wäre es wohl Sinnvoll eine Lizenz zu kaufen...
Leider hängte ich immer noch etwas bei der EInrichtung sorry...
ctl_fifo.h kann auch in CrossWorks nicht gefunden werden?

sarnau, 2019-07-07:
Ahh, guter Punkt. Ist aber auch verfügbar: https://github.com/sarnau/ctl_fifo

challo2018, 2019-07-08:
Danke!
Jetzt habe ich noch in der system.cpp den Fehler:
'ctl_set_priority' was not declared in this scope

Hast du hier auch eine Idee?

challo2018, 2019-07-08:
Ich habe noch folgenden Fehler, hast du eine Idee?
Output/Debug/Exe/LoxLink.elf does not exist.
no linker script, memory map file or section placement segments specified

sarnau, 2019-07-08:
Du hast die STM32 libraries nicht installiert, von dort kommt ctl_set_priority(). Ohne die Funktion läuft de Linker natürlich nicht durch,

challo2018, 2019-07-08:
ctl_set_priority() habe ich gelöst, sorry hatte den Kommentar dazu schon gelöscht. Jetzt habe ich nur noch ein linker Fehler... Siehe oben

sarnau, 2019-07-08:
Hast Du den CPU Type 100% correct eingestellt? Das erstmalige Setup kann leider dauern :-(

challo2018, 2019-07-09:
Hab das Projektfile von dir in segger embedde Studio verwendet. Libraries etc. habe ich alle aus crossworks verknüpft. package für stm32f1xx hab ich installiert und controller ist in den optionen wie in deiner config gewählt, das wurde aus dem projektfile übernommen.

sarnau, 2020-01-09:
Hmm, gute Frage. ctl_…() ist Standard-Code von CrossStudio. Der Source ist sogar dabei, da müßte sich das finden lassen?!?

matlab22, 2020-01-12:
Habe mir nun die Testversion von CrossStudio heruntergeladen um zu schauen, ob es dort funktioniert.
Kann den code nun compilieren, jedoch nicht auf mein device laden.
DEV_AP_ACCESS_ERROR
[image]

sarnau, 2020-01-12:
Nichtmal Google kennt diese Meldung. Ich benutze einen J-Link PRO von Segger für den Upload auf den STM32 – habe das Interface vor einiger Zeit mal sehr günstig bei eBay gekauft.

Die 2 Euro Adapter von Aliexpress funktionieren aber auch – mit der richtigen Firmware.

chrisrock1984, 2020-01-12:
Hi, verwende das Blue Pill Board mit einem STM32F103C8T6 sowie die gratis Version von Crossstudio (mit 16Kb Limit). CPU habe ich umgestellt, kompiliert (DI Extension) und geflasht. Leider funktionierts am LoxLink nicht. Sobald ich das Board einschalte erkennt Loxone die vorhandenen Extensions nicht mehr. Verkabelung, CAN Pins, DI Pins wurden kontrolliert bzw. im Code angepasst. Ich habe die Vermutung, dass beim Clock etwas nicht richtig konfiguriert ist. Leider kenne ich mich beim Crossstudio auch zu wenig aus. Vielleicht hat jemand schon das tolle Projekt auf den STM32F103C8T6 portiert und könnte helfen ;-)

Danke!!

matlab22, 2020-01-12:
> 
> 
> Nichtmal Google kennt diese Meldung. Ich benutze einen J-Link PRO von Segger für den Upload auf den STM32 – habe das Interface vor einiger Zeit mal sehr günstig bei eBay gekauft.
> 
> Die 2 Euro Adapter von Aliexpress funktionieren aber auch – mit der richtigen Firmware.

Ich habe nun mit CubeMX ein EWARM Projekt mit den beiden LEDs an PB13&14 gemacht und dieses in CrossStudio importieren lassen. Dort lasse ich die LEDs abwechselnd blinken. Ich kann das Miniprojekt problemlos mit meinem STLink2 oder dem J-Link clone laden.
Es liegt demnach an irgend einer Einstellung im LoxLink Projekt.

chrisrock1984, 2020-01-12:
Bei EmbeddedStudio bekomme ich folgenden Fehler -> ctl_api.h not found.

matlab22, 2020-01-12:
> Hi, verwende das Blue Pill Board mit einem STM32F103C8T6 sowie die gratis Version von Crossstudio (mit 16Kb Limit). CPU habe ich umgestellt, kompiliert (DI Extension) und geflasht. Leider [...]

Hast du die beiden LEDs in LED.cpp auch auf dein Board umgemappt? Was machen die wenn du das Ding startest?

chrisrock1984, 2020-01-12:
> Hast du die beiden LEDs in LED.cpp auch auf dein Board umgemappt? Was machen die wenn du das Ding startest?

Guter Punkt - dort habe ich nichts geändert. Das Board hat nur eine ansteuerbare LED.

matlab22, 2020-01-12:
> 
> 
> Bei EmbeddedStudio bekomme ich folgenden Fehler -> ctl_api.h not found.

Hast du die Packages installiert?
[image]

chrisrock1984, 2020-01-12:
> > Bei EmbeddedStudio bekomme ich folgenden Fehler -> ctl_api.h not found.
> 
> Hast du die Packages installiert?
> [image]

ich wollte nur zum testen das Projekt im Segger EmbeddedStudio kompilieren. Im Crossstudio funktioniert das kompilieren ohne Probleme. Im Segger Package Manager findet er diese Library nicht mal.

matlab22, 2020-01-12:
@chrisrock1984 
Das Problem hatte ich mit Segger Studio auch. Konnte dies lösen, indem ich $(TargetDir) von Hand verlinkt hatte.

chrisrock1984, 2020-01-12:
> @chrisrock1984
> Das Problem hatte ich mit Segger Studio auch. Konnte dies lösen, indem ich $(TargetDir) von Hand verlinkt hatte.

Danke für den Hinweis. Leider bin ich kein EmbeddedStudio Profi und weiß nicht wie man das macht.

matlab22, 2020-01-12:
Ich weiss auch nicht was ich tu ;)
Aber alle diese Files müssen vorhanden sein. Schau mal mit doppelklick, ob die Dateien alle verlinkt sind.
[image]

chrisrock1984, 2020-01-12:
> Ich weiss auch nicht was ich tu ;)
> Aber alle diese Files müssen vorhanden sein. Schau mal mit doppelklick, ob die Dateien alle verlinkt sind.
> [image]

Dankeschön. Leider habe ich es nicht geschafft alle System Dateien zu verlinken. Konzentriere mich auf Crossstudio, dort funktioniert das kompilieren. Habe auch die Boards mit dem richtigen STM32 bestellt da das Blue Pill Board nicht so tut wie es soll. Ich werde es weiterhin versuchen, da diese Boards bereits im < 2€ Bereich erhältlich sind.

Jedenfalls nochmal Danke für jede Hilfe!!

Lg

challo2018, 2020-01-20:
Hab in der Zwischenzeit ein STM32F103VET6 .
In Crossstudio bekomme ich beim Flashen mit ST Link v2 -> DEV_AP_ACCESS_ERROR
hab daraufhin auch mit segger j-link plus getestet. -> Dort bekomme ich den Fehler Can't read Memory. 
@matlab22 : Mit welchem Programmer und welchem Board läuft es bei dir? Segger oder Crossstudio?

matlab22, 2020-01-20:
Genau die gleichen Fehler hatte ich auch. Was passiert, wenn du mit dem Segger verbunden bist und auf Reset klickst? Bei mir ist dort der Segger auch abgeschmiert.
In der Zwischenzeit laufen bei mir beide JLink und STLink2. Ich kann dir jedoch nicht wirklich sagen, weshalb es nun funktioniert. Ich hatte STM32CubeMX heruntergeladen und ein kleines EWARM Projekt mit den beiden LEDs gemacht. Dies anschliessend in CrossStudio importiert. Das hat dann auf anhieb geklappt. Das Loxlink nach wie vor nicht.
Ich habe danach in CrossStudio ein neues Projekt CrossWorks Tasking Libary executable for ST... erstellt. Dies hat auch funktioniert. Das LoxLink nach wie vor nicht. Darauf hin wollte ich untersuchen wie sich die beiden Project files unterscheiden. Bin aber irgendwie nicht wirklich schlau geworden. Hab dann irgendwie etwas aufgegeben. Ein paar Tage später wollte ich mich noch einmal darum kümmern und das LoxLink Projekt hat auf anhieb geklappt. Seit dem funktionierts einwand frei.... Eventuell hat der Neustart geholfen ...

sarnau, 2020-01-20:
Kann jemand Python nach C übertragen? Ich bin nicht sicher, ob ich Zeit dafür habe die Autorisierung nach C zu übertragen und zu testen.

matlab22, 2020-01-20:
> 
> 
> Kann jemand Python nach C übertragen? Ich bin nicht sicher, ob ich Zeit dafür habe die Autorisierung nach C zu übertragen und zu testen.

Ich kann es zumindest versuchen ;)

sarnau, 2020-01-20:
Wir brauchen eine 128-bit AES CBC Encryption und Decryption. Habe noch nicht gesucht, eine minimale AES Lib mit BSD Lizenz sollte sich finden lassen.

Dazu noch RSHash, JSHash, DJBHash, DEKHash und BPHash (ja, ist albern und trivial).

matlab22, 2020-01-20:
Wie ist das ganze in loxone aufgebaut? Was wird gemacht um die extensions zu verifizieren?

AES CBC könnte uns diese lib liefern:
https://github.com/kokke/tiny-AES-c/blob/master/README.md

sarnau, 2020-01-20:
Miniserver sendet eine Challenge (ein verschlüsselte Block mit einer Zufallszahl drin) und die Extension/das Device muß diese Challenge entschlüsseln und eine entsprechende Antwort zurückschicken – die AES-Hauptschlüssel sind sowohl im Miniserver, wie auch in jedem Gerät enthalten. Das genaue Verfahren ist zwar extrem umständlich und gleichzeitig nicht besonders sicher.

Auch interssant: AI und AO Extension sind noch nicht angepaßt, die habe noch kein Challenge/Response Verfahren! Ob es notwendig ist, kann man an der Versionsnummer der Extensions/Devices erkennen. Diverse legacy Extensions habe anscheinend auch noch kein Update bekommen – sie schützen hauptsächlich die Extension und die Relay Extension. Was albern ist, denn 8 Relays kann man natürlich auch via UDP ansteuern, wenn man denn muß…

Ehrlich gesagt: was mich nervt, ist daß Loxone zwar verständlicherweise Ihre eigene Hardware verkaufen will, es aber gleichzeitig sehr umständlich/unmöglich macht eine gute Einbindung von exotischer Hardware zu ermöglich. Auch Hobbyisten, welche der Firma nur wirklich nicht schaden, werden gnadenlos ausgesperrt. Das ist bei keinem anderen System so (KNX, etc), außer vielleicht bei HomeKit (wobei es dort inzwischen eine öffentliche Doku gibt). Besser wäre es den ganzen Kram zu dokumentieren und bei kommerzielle Nutzung eine Lizenz zu verlangen. Das macht das System sicher und es ist sogar einfacher für Loxone.

sarnau, 2020-01-25:
Authorization Support ist drin. Für weitere Diskussionen bitte eine neue Issue aufmachen!

closed 2020-01-25 by sarnau

chrisrock1984, 2020-02-03:
> @chrisrock1984
> Das Problem hatte ich mit Segger Studio auch. Konnte dies lösen, indem ich $(TargetDir) von Hand verlinkt hatte.

Wie setzt man das TargetDir? Habe diese Funktion nicht gefunden.