sarnau/LoxLink #2: Di Extension booted nicht

matlab22, 2020-01-17:
Ich habe main.cpp angepasst, so dass mein [STM32F103VET6 ](https://www.aliexpress.com/item/32693281117.html?spm=a2g0s.9042311.0.0.27424c4dVrkkHq) Board eine DiExtension werden soll:

```
#include "system.hpp"

#include "LED.hpp"
#include "Watchdog.hpp"

#include "LoxCANDriver_STM32.hpp"
//#include "LoxBusTreeExtension.hpp"
#include "LoxBusDIExtension.hpp"
//#include "LoxBusTreeAlarmSiren.hpp"
//#include "LoxBusTreeRoomComfortSensor.hpp"
//#include "LoxBusTreeTouch.hpp"
//#include "LoxLegacyDMXExtension.hpp"
//#include "LoxLegacyRS232Extension.hpp"
//#include "LoxLegacyModbusExtension.hpp"
//#include "LoxLegacyRelayExtension.hpp"
//#include "LoxBusTreeRgbwDimmer.hpp"

int main(void) {
  system_init();

  static CTL_TASK_t main_task;
  ctl_task_init(&main_task, 255, "main"); // create subsequent tasks whilst running at the highest priority.

  // Warning: be aware that two relay extension need two different serial numbers!
  uint32_t serial_base = serialnumber_24bit();

  static LoxCANDriver_STM32 gLoxCANDriver(tLoxCANDriverType_LoxoneLink);
#if EXTENSION_RS232
//  static LoxLegacyRS232Extension gLoxLegacyRS232Extension(gLoxCANDriver, serial_base);
#endif
#if EXTENSION_MODBUS
//  static LoxLegacyModbusExtension gLoxLegacyModbusExtension(gLoxCANDriver, serial_base);
#endif
  static LoxBusDIExtension gDIExtension(gLoxCANDriver, serial_base, gResetReason);
  //  static LoxLegacyRelayExtension gRelayExtension(gLoxCANDriver, serial_base);
  //  static LoxLegacyDMXExtension gDMXExtension(gLoxCANDriver, serial_base);
  //  static LoxBusTreeExtension gTreeExtension(gLoxCANDriver, serial_base, gResetReason);
  //  static LoxBusTreeRoomComfortSensor gTreeRoomComfortSensor(gTreeExtension.Driver(eTreeBranch_rightBranch), 0xb0112233, gResetReason);
  //  gTreeExtension.AddDevice(&gTreeRoomComfortSensor, eTreeBranch_rightBranch);
  //  static LoxBusTreeTouch gLoxBusTreeTouch(gTreeExtension.Driver(eTreeBranch_leftBranch), 0xb010035b, gResetReason);
  //  gTreeExtension.AddDevice(&gLoxBusTreeTouch, eTreeBranch_leftBranch);
  //  static LoxBusTreeAlarmSiren gLoxBusTreeAlarmSiren(gTreeExtension.Driver(eTreeBranch_leftBranch), 0xb010035c, gResetReason);
  //  gTreeExtension.AddDevice(&gLoxBusTreeAlarmSiren, eTreeBranch_leftBranch);
  //  static LoxBusTreeRgbwDimmer gLoxBusTreeRgbwDimmer(gTreeExtension.Driver(eTreeBranch_rightBranch), 0xb0200000, gResetReason);
  //  gTreeExtension.AddDevice(&gLoxBusTreeRgbwDimmer, eTreeBranch_rightBranch);

#if DEBUG && 0
  MX_print_cpu_info();
#endif
  gLED.Startup();
  gLoxCANDriver.Startup();

  Start_Watchdog();
  ctl_task_set_priority(&main_task, 0); // drop to lowest priority to start created tasks running.
  while (1) {
  }
  return 0;
}
```
Ich habe die BuildConfig auf Release gestellt. Dadurch musste ich in LoxBusTreeRgbwDimmer.cpp
```
void LoxBusTreeRgbwDimmer::ReceiveDirect(LoxCanMessage &message) {
    **//message.print(this->driver);**
    LoxBusTreeDevice::ReceiveDirect(message);
}
```
message.print(this->driver) auskommentieren, da sonst ein compiler Fehler kommt:
'class LoxCanMessage' has no member named 'print'

LoxCANBaseDriver.cpp
```
/***
 *  Received a message
 ***/
void LoxCANBaseDriver::ReceiveMessage(LoxCanMessage &message) {
  ++this->statistics.Rcv;
#if DEBUG && 0
  debug_printf("CANR:");
  **message.print(*this);**
#endif
  for (int i = 0; i < this->extensionCount; ++i) {
    this->extensions[i]->ReceiveMessage(message);
  }
}

```

Das eigentliche Problem was ich nun habe ist, dass die beiden LEDs D3 und D2 nur gerade nach dem uploaden des codes zu blinken beginnen. Wenn ich den Chip resette über SW1 oder CrossStudio blinken diese nicht mehr. Habe BOOT0 und BOOT1 auf 0. An was liegt das weshalb started der Code nur gerade nach dem Flashen?

sarnau, 2020-01-19:
Ich habe noch nie was mit BOOT0/BOOT1 gemacht – es funktionierte bei mir einfach so.

sarnau, 2020-01-19:
Funktioniert ein einfach "Blink-LED" Programm?

matlab22, 2020-01-19:
Guter Punkt. Habe festgestellt, dass mein einfaches Blinky nach einem Reset auch nicht startet. Habe nun in CrossStudio STARTUP_FROM_RESET im Preprocessor definiert.
[image]
Nun startet das Ding auch jedesmal auf.
Aber so wie ich [diesen Beitrag](https://rowley.zendesk.com/hc/en-us/articles/210032923-Application-not-starting-up-from-reset) interpretiere sollte ich STARTUP_FROM_RESET nicht definieren.

sarnau, 2020-01-19:
Super, der Setup – im Vergleich zum Arduino – ist beim STM32 wirklich nervig. Wenn es aber mal läuft ist es deutlich besser, gerade im Hinblick auf ordentliches Debugging!

closed 2020-01-19 by matlab22