sarnau/LoxLink #3: Authentifikation fehlgeschlagen

matlab22, 2020-01-19:
Ich habe es nun geschafft die Di Extension in Betrieb zu nehmen. Die Freude war aber von sehr kurzer Dauer. Loxone wollte die Extension von auf 10.02.02.28 auf 10.03.11.08 aktualisieren. Dies ist logischerweise fehlgeschlagen. Damit er diese nicht aktualisieren will habe ich die Versionsnummer in LoxBusDIExtension.cpp auf 10031108 geändert. Worauf hin Loxone einene Meldung gebracht hat und die Extension d3eaktivert hat. 
[image]
Vorher hatte ich noch kurz einen Eingang unter der alten Versionsnummer testen können :)

Hast du einen Tipp wie man diese Authentifikation vorgaukeln kann?

sarnau, 2020-01-24:
Ok, aktuell mußt Du 10.2 nutzen, nicht die 10.3. Dann funktionieren alle Extensions und Devices.

sarnau, 2020-01-25:
Ich habe gerade Support für Authorization gepusht.

closed 2020-01-25 by sarnau