sarnau/LoxLink #4: RS232 Extension -> queue.h fehlt

matlab22, 2020-01-26:
Ich konnte erfolgreich eine DiExtension in Betrieb nehmen. Vielen Dank für die Kryptoerweiterung 👍 

Nun wollte ich die RS232 ausprobieren und habe dazu in LoxLegacyExtension.hpp
'''
#define EXTENSION_RS232 1
'''
von 0 auf 1 gewechselt.

Jetzt reklamiert er jedoch in LoxLegacyRS232Extension.hpp, dass er Zeile 13 #include "queue.h" nicht finden kann. Wo finde ich diese Datei?

sarnau, 2020-01-26:
Das sind Reste von meiner alten FreeRTOS Version der Extensions. Ich habe die RS232 Extension und die Modbus Extension noch nicht an CrossStudios CTL angepaßt.

matlab22, 2020-01-26:
Ich habe gerade die Relais Extension erfolgreich bei mir integrieren können. Habe anschliessend die DMX in Betrieb genommen. Loxone wollte diese aktualisieren. Habe nun die neue Versionsnummer hinterlegt und das scheint auszureichen. Authentifizierung ist scheinbar hier nicht nötig :)
Werde mir bei Gelegenheit die RS232 nach vorlage der DMX anpassen. Wenn das Ding funktioniert würde ich diese dann pushen. Gehe nächste Woche Skifahren. Je nach Wetter wird jedoch an der Extension gebastelt 💃

sarnau, 2020-01-26:
Super, danke Dir! Ja, die Versionsnummer muß angepaßt werden – aber nicht alle Extensions brauchen die Autorisierung. Macht aber nix.

matlab22, 2020-01-27:
Hat die RS232 bei dir mal funktioniert? Ich habe mal versuchshalber den [d295489](https://github.com/sarnau/LoxLink/tree/d295489086ec79da9d6af7f65d1fec67932755bc) ausprobiert. Das sollte der letzte Stand sein bevor du auf CTL gewechselt hast. Kann es sein dass die RS232 auf einem anderen Target lief?

sarnau, 2020-01-27:
Ja, ich habe die RS232 damals durchgetestet. Wobei die RS232 mit Loxone ist schon etwas eigen. Ich sehe nur sehr wenige Anwändungsfälle.

matlab22, 2020-02-01:
Habe die freertos version zum laufen bekommen. Scheinbar ist eines meiner Boards defekt. Auch andere Extensions laufen damit nicht. Werde mal den TJA1050 ersetzen und noch mal probieren.
PS. RS232 ist gemäss Loxone auf Firmware Version: 10031108.

matlab22, 2020-02-01:
Kriegs irgendwie doch nicht hin. Habe mit einem USB zu RS232 und HTerm versucht gekreuzt und ungekreuzt zu kommunizieren. Beim Empfangen ist nichts im RS232 Monitor von Loxone und beim senden Befehl bei EIN \x04 kommt auch nichts am HTerm an.
Wie hattest du dies getestet? Ich würde es gerne zum laufen bringen bevor ich mich an die Portierung mache.

MS-Astra, 2021-02-23:
Moin, habt ihr Lust, an der Baustelle gemeinsam noch etwas weiter zu arbeiten?