sarnau/LoxLink #6: DI Extension booted nur mit ST-Link Utility

chrisrock1984, 2020-01-30:
Hallo,

gleich vorweg: Ja, ich habe "STARTUP_AFTER_RESET" gesetzt. Leider ist der Erfolg nur halb eingetreten.
Die Extension bootet nur wenn ich im ST-LINK Utility unter Target->MCU Core-> System Reset oder Core Reset ausführe (die LED beginnt zu blinken). Boot0/1 sind beide auf GND.

Im Code selber habe ich nur die grüne LED auf PC13 gehängt.

Was mache ich falsch, dass der STM32 nicht alleine bootet?

Danke!!!

chrisrock1984, 2020-01-30:
Um zu sehen, ob das Board etwas hat, habe ich einen Bootloader drauf geflasht. Dieser hat ohne Probleme funktioniert. Beim Flash Tool habe ich keine Einstellungen vornehmen müssen. Bleibt nur mehr Crossstudio übrig als Ursache, oder?

matlab22, 2020-02-01:
Schick mal dein hex, ich schau ob und was anderst ist gegenüber meinem.

chrisrock1984, 2020-02-02:
Bitteschön!! und DANKE!!
[LoxLink.zip](https://github.com/sarnau/LoxLink/files/4143726/LoxLink.zip)

matlab22, 2020-02-02:
Hmm deine Hex sieht sehr anders aus. Hast du auch ein Board mit STM32F103 und externem Quarz? Weshalb hast du auf PC13 gewechselt?

chrisrock1984, 2020-02-02:
8MHz Quarz vorhanden
Plus 32khz quarz
Es ist ein stm32f103vet6 drauf
PC13 ist die on board led, die ich verwende als grüne extension led.

chrisrock1984, 2020-02-02:
https://images.app.goo.gl/upT8qa44oGLMo2Fy7

matlab22, 2020-02-02:
Wie wird das Board gespiesen? Ich sehe kein 5V zu 3.3V Regler auf dem Board?
Wie programmierst du den Chip? STM32CubeProgrammer?

Teste mal dieses File:
[STM32F103VE_CAN_Board.zip](https://github.com/sarnau/LoxLink/files/4144156/STM32F103VE_CAN_Board.zip)
Pin 13 und 14 sollten abwechslungsweise getoggelt werden.
`  while (1)
  {
	  HAL_GPIO_TogglePin(GPIOC, GPIO_PIN_13);
	  HAL_Delay(100);
	  HAL_GPIO_TogglePin(GPIOC, GPIO_PIN_13);
	  HAL_GPIO_TogglePin(GPIOC, GPIO_PIN_14);
	  HAL_Delay(500);
	  HAL_GPIO_TogglePin(GPIOC, GPIO_PIN_14);
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }`

chrisrock1984, 2020-02-02:
Der regler ist unterhalb
Programmiert wird es über SWD

Ok

chrisrock1984, 2020-02-02:
Mit dem ST-Link utility als Tool

matlab22, 2020-02-02:
Dein Hex startet übrigens auf meinem Board. Aber PC13 bleibt dunkel.

Hast du die Portinitialisierung auch angepasst?
```
void LED::Startup(void) {
  __HAL_RCC_GPIOB_CLK_ENABLE();

  GPIO_InitTypeDef GPIO_Init;
  // 2 LEDs as outputs
  GPIO_Init.Pin = GPIO_PIN_13 | GPIO_PIN_14;
  GPIO_Init.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_Init.Pull = GPIO_NOPULL;
  GPIO_Init.Speed = GPIO_SPEED_FREQ_LOW;
  HAL_GPIO_Init(GPIOB, &GPIO_Init);

  // 2 buttons as inputs
  GPIO_Init.Pin = GPIO_PIN_1 | GPIO_PIN_15;
  GPIO_Init.Mode = GPIO_MODE_INPUT;
  GPIO_Init.Pull = GPIO_PULLDOWN;
  HAL_GPIO_Init(GPIOB, &GPIO_Init);

  #define STACKSIZE 128          
  static unsigned stack[1+STACKSIZE+1];
  static CTL_TASK_t led_task;
  ctl_task_run(&led_task, 2, LED::vLEDTask, this, "LED", STACKSIZE, stack+1, 0);
}```

chrisrock1984, 2020-02-02:
mein LED Code 👍 
void LED::Startup(void) {
  __HAL_RCC_GPIOB_CLK_ENABLE();

  GPIO_InitTypeDef GPIO_Init;
  // 2 LEDs as outputs
  GPIO_Init.Pin = GPIO_PIN_14;
  GPIO_Init.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_Init.Pull = GPIO_NOPULL;
  GPIO_Init.Speed = GPIO_SPEED_FREQ_LOW;
  HAL_GPIO_Init(GPIOB, &GPIO_Init);

  GPIO_Init.Pin = GPIO_PIN_13;
  GPIO_Init.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_Init.Pull = GPIO_NOPULL;
  GPIO_Init.Speed = GPIO_SPEED_FREQ_LOW;
  HAL_GPIO_Init(GPIOC, &GPIO_Init);

chrisrock1984, 2020-02-02:
leider kann ich mit meinem ST-LINK nicht debuggen - bekomme einen Fehler. Deshalb weiß ich auch nicht ob das Board startet. Die LED beginnt zu blinken wenn ich wie im ersten Post oben einen Core-Reset im ST-Link utility ausführe.

sarnau, 2020-02-02:
Google mal rum: man kann ein ST-Link auf einen J-Link umflashen. Zumindest mit dem J-Link funktioniert debugging mit Breakpoints, etc. – so wie es soll.

chrisrock1984, 2020-02-02:
> Wie wird das Board gespiesen? Ich sehe kein 5V zu 3.3V Regler auf dem Board?
> Wie programmierst du den Chip? STM32CubeProgrammer?
>
> Teste mal dieses File: [...]

Kann das elf File leider nicht flashen. STM32CubeProgrammer meldet einen Fehler - invalid ELF.
Mit ST-LINK utility kann ich nur hex Files flashen

matlab22, 2020-02-02:
[STM32F103VE_CAN_Board.zip](https://github.com/sarnau/LoxLink/files/4144194/STM32F103VE_CAN_Board.zip)

hier noch das hex, welches ich ursprünglich anhängen wollte ;)

chrisrock1984, 2020-02-02:
> Google mal rum: man kann ein ST-Link auf einen J-Link umflashen. Zumindest mit dem J-Link funktioniert debugging mit Breakpoints, etc. – so wie es soll.

Danke.
Diesen hier auch? ->https://www.google.com/url?sa=i&url=https%3A%2F%2Ftecnoton.net%2Fproducto%2Fst-link-stlink-st-link-v2-mini-stm8-stm32-simulator-download-programmer-programming-with-cover-dupont-cable%2F&psig=AOvVaw0kml71rvIFd53s8H-ExpJL&ust=1580735176787000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCICVnsf3sucCFQAAAAAdAAAAABAI

chrisrock1984, 2020-02-02:
> [STM32F103VE_CAN_Board.zip](https://github.com/sarnau/LoxLink/files/4144194/STM32F103VE_CAN_Board.zip)
> 
> hier noch das hex, welches ich ursprünglich anhängen wollte ;)

Jep - LED auf PC13 blinkt

chrisrock1984, 2020-02-02:
was mache ich bein Compilieren falsch? ;-/

matlab22, 2020-02-02:
ich vermute dein Code ist nicht richtig angepasst.
```
/***
 *   Set the LED gpio
 ***/
static void LED_on_off(eLED_color color) {
  switch (color) {
  case eLED_green: // right LED on
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_13, GPIO_PIN_SET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_RESET);
    break;
  case eLED_orange: // left LED on
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_13, GPIO_PIN_RESET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_SET);
    break;
  case eLED_red: // both LEDs on
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_13, GPIO_PIN_RESET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_RESET);
    break;
  default: // both LEDs off
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_13, GPIO_PIN_SET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_SET);
  }
}
```
Hast du hier auch von B auf C umgestellt?

chrisrock1984, 2020-02-02:
> ich vermute dein Code ist nicht richtig angepasst.
>
> ```
> /***
>  *   Set the LED gpio
>  ***/
> static void LED_on_off(eLED_color color) {
>   switch (color) {
>   case eLED_green: // right LED on [...]

Klar doch - die LED blinkt ja, wenn ST-Link angeschlossen und per SW ein Core Reset ausgeführt wird. Starte ich das Board alleine blinkt nichts mehr.

chrisrock1984, 2020-02-02:
/***
 *   Set the LED gpio
 ***/
static void LED_on_off(eLED_color color) {
  switch (color) {
  case eLED_green: // right LED on
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_RESET);
    break;
  case eLED_orange: // left LED on
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_SET);
    break;
  case eLED_red: // both LEDs on
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_RESET);
    break;
  default: // both LEDs off
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_SET);
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_14, GPIO_PIN_SET);
  }
}

matlab22, 2020-02-02:
Hast du auch einen Clock Enable für GPIOC gemacht?
`__HAL_RCC_GPIOB_CLK_ENABLE(); `
Dein hex file blinkt bei mir auf PC14 wenn ich es aus und wieder einstecke

chrisrock1984, 2020-02-02:
> Hast du auch einen Clock Enable für GPIOC gemacht?
> `__HAL_RCC_GPIOB_CLK_ENABLE(); `
> Dein hex file blinkt bei mir auf PC14 wenn ich es aus und wieder einstecke

nein, anscheinend nicht. wo muss ich das machen?

chrisrock1984, 2020-02-02:
habs schon gesehen ;-)

chrisrock1984, 2020-02-02:
LED Blinkt jetzt ;-). Mal schaun ob Loxone was findet!! 
Vielen Dank schon mal!!!

chrisrock1984, 2020-02-02:
JUHU!!! Loxone findet die DI Extension!!!!!!

Vielen Lieben Danke für eure Tipps und vor allen Zeit!!!

matlab22, 2020-02-02:
kein Problem, pass doch noch den Titel an und schliess das Issue

closed 2020-02-02 by chrisrock1984