sarnau/LoxLink #7: DMX Extension Authentication Fail

matlab22, 2020-02-01:
Sobald ich der DMX Extension Aktoren zuweise kommt das Autentifizierungsproblem wieder :(
Kann es sein, dass die Legacy sich doch etwas anders autentifizeren?
Habe ich es falsch verstanden, dass LoxLegacyDMXExtension von der LoxLegacyExtension erbt und sich die Methode FragmentedPacketToExtension um den Crypto-Key kümmert? Muss ich noch etwas anderes beachten, dass ich eventuell übersehen habe?

sarnau, 2020-02-01:
Ja, tun sie. Der Algorithmus ist völlig gleich, wird aber via dem Legacy-Fragmented-Package übertragen. Ich habe den DMX Code noch nicht selbst ausprobiert, habe es nur mit der Relay-Extension ausprobiert.

chrisrock1984, 2020-02-02:
> Sobald ich der DMX Extension Aktoren zuweise kommt das Autentifizierungsproblem wieder :(
> Kann es sein, dass die Legacy sich doch etwas anders autentifizeren?
> Habe ich es falsch verstanden, dass [...]

Welche Pins werden für den RS485 Transceiver verwendet? Im Code habe ich leider nichts gefunden, oder wahrscheinlich übersehen.

sarnau, 2020-02-02:
Kann mich nicht mehr erinnern, aber RS485 wird wie RS232 angesteuert. Ist quasi identisch.

chrisrock1984, 2020-02-04:
Eine andere Frage - Ist die Implementierung der DMX Extension auch funktional? Ich habe mir den Code der Modbus sowie der RS232 Extension angesehen. Dort findet sich Code für den HW Zugriff. Dei der DMX Implementierung habe ich das nicht gefunden.
Bitte um Richtigstellung ;-)

Danke!!

sarnau, 2020-02-04:
Nein, ich habe nicht alle Extensions und Geräte nachprogrammiert. Gerade DMX braucht ja eine spezielle RS485 Ansteuerung mit fixem Timing.

chrisrock1984, 2020-02-04:
Gut ;) Ich kann noch Code lesen! Danke für die Info!

closed 2020-03-29 by sarnau