sarnau/LoxLink #9: Segger Embedded Studio for Arm

chrisrock1984, 2020-02-03:
Hallo,

Habe versucht das Crossstudio Projekt nach Segger zu portieren. Leider bekomme ich ne Menge Linker Errors. Liegt wahrscheinlich daran, dass Crossstudio Datein benötigt werden. 

Hat irgendjemand diese Portierung bereits erledigt?
Andere Alternative, die frei verfügbar ist?
Ich würde gerne, nach Ablauf der Evaluation Phase weiter am Code herumbasteln.

Danke!!

sarnau, 2020-02-04:
Die "CrossWorks Tasking Library" ist notwendig. Die kompiliert zwar auch unter Seggers (schließlich ist Segger nur eine rebranded Version von Crossworks), aber gehört soweit ich weiß, nicht dazu. Ich hatte unsprünglich mal FreeRTOS genommen, aber hatte ein paar Probleme und bin dann – wie Loxone anscheinend auch – auf die CTL gegangen.

KjellVerb, 2020-04-19:
Hi, I understand German but writing is another thing... I'll answer in English if you don't mind.

@chrisrock1984 , did you make any progress porting this project to Segger? I'm stuck and making no progress on my own. 

I extracted the missing "System Files" from the free version of CrossStudio and hand-linked the missings. I also added the environment variable "$(TargetDir)=/Users/...".
Now I'm still facing some issues, one of them being that I can't find where to define the target to be used (#error target not defined). Full disclosure: I'm still waiting for my ST-link V2 to arrive so currently I'm trying to build without connecting to the STM32.

chrisrock1984, 2020-04-19:
Hi,

Did not make any progress so far ;(

Br

KjellVerb, 2020-04-19:
Apparently I needed to restart Segger in order to load the environment variable, now I got a bit further.

I had the "ctl_set_priority()"-error and solved it (I think) by adding "#define __CROSSWORKS_ARM".
Now the project is completely building but fails at the end with the following error: "no linker script, memory map file or section placement segments specified" @sarnau , I'm not experienced in STM32 development so could you point me in the right direction please?

sarnau, 2020-04-19:
That is the hardest part of getting the STM32 stuff running. I would suggest to create a trivial STM32 example with Segger and compare the project file with the crossworks one. They are almost identical, so you should be able to find what you need to adjust.

KjellVerb, 2020-04-19:
Thanks for the hint, I think my setup is all OK now because I'm getting no more errors and the build completes. Now the real test will be when I get all the hardware for an actual setup.

chrisrock1984, 2020-05-02:
Is it possible to share the segger project files as I already have all the hardware ?!
Really appreciate your work you have done!!

Thx

KjellVerb, 2020-06-09:
My hardware has arrived but unfortunately I'm still struggling.
I'm able to start and stop the program with the debugger, but it won't run on it's own. This seems like another one of those weird STM32 things. Any suggestions on where to look? 

I'll try to retrace and write down my steps once everything is up.

KjellVerb, 2020-06-09:
Turns out that the code crashed on `debug_printf()` :-)

KjellVerb, 2020-06-09:
Another question that may or may not be related to the Segger porting: 
I'm currently debugging with an endless blinky loop running in LED.cpp. It seems like `HAL_Delay()` has an LSB delay value of 10ms instead of 1ms, is that expected behaviour in the LoxLink project or do I need more debugging to rectify this? `system_init()` has already been called so I do think that all of the config should have been done by the time I get into my infinite loop.

KjellVerb, 2020-06-11:
Never mind that last question, I finally have a loxlink extension up and running on two different STM32 eval-boards. I'll try to retrace my steps in the coming days and will try to document the process.

chrisrock1984, 2020-06-11:
That would be great!!

challo2018, 2020-09-03:
@KjellVerb Is it possible that you share your running segger studio project?

chrisrock1984, 2020-09-03:
> @KjellVerb Is it possible that you share your running segger studio project?

That would be awesome ;)!!!

KjellVerb, 2020-09-04:
I'm not very eager to share the full project folder because
1) it contains Crossworks libraries that I am not _allowed_ to distribute
2) it will not help you because you also need to change some other folders on your drive
3) there are OSX specific references in the project

What I _can_ share is the process of how I got it to work. I restarted from scratch and logged my steps, see attached.
[LoxLink2Segger.txt](https://github.com/sarnau/LoxLink/files/5172895/LoxLink2Segger.txt)

chrisrock1984, 2020-09-04:
> I'm not very eager to share the full project folder because
>
> 1. it contains Crossworks libraries that I am not _allowed_ to distribute
> 2. it will not help you because you also need to change some [...]

GREAT!! works for me ;-) was able to compile. now I need to check if it works on silicon ;-)
Thanks a lot for your work!!

KjellVerb, 2020-09-04:
You're welcome!
@sarnau I think this one can be closed.

dimmu311, 2020-11-15:
Hi, i also try to compile with segger. after spending one day of trying to port from crossstudio to segger i fund this process here. i managed everything right down to the last to steps.
now if i try to compile it stops. what i found out for now is that i have not specified target processor. i don't know where and how to specify the project options. could anyone please help? Thank you

davidwallis3101, 2021-01-21:
just trying to follow these instructions using segger emedded studio for arm v5.34 64bit on windows.. 

I'm getting the following errors linking LoxLink.elf

```
Checking project status
Building ‘LoxLink’ from solution ‘LoxLink’ in configuration ‘Debug’
  Linking LoxLink.elf
    Output/Debug/Obj/LoxLink/STM32_ctl.o: in function `SysTick_Handler':
    undefined reference to `ctl_enter_isr'
    C:\Users\Davidw\AppData\Local\SEGGER\SEGGER Embedded Studio\v3\packages\targets/STM32/STM32_ctl.c:60: undefined reference to `ctl_exit_isr'
Build failed
```

Struggling a little with the config instructions possibly:

		- Correct memory map file
		- Correct DeviceVectorsFile under Project Macros
		- Correct target processor
Not sure what I need to do with these when using a STM32103VET..

Added #define __CROSSWORKS_ARM to system.cpp - correct?
TargetsDir I've sorted.. 

Any suggestions welcome :)

dimmu311, 2021-01-21:
i am not sure if you struggling at the same point i did.
i am also not sure if it is the easyest way i did, but this way it worked for me.

see this picture how i added _CROSSWORKS_ARM.

[image]

davidwallis3101, 2021-01-21:
Bingo! That compiled.. 

Now need to to work out how to debug and program with the st-link v2

MS-Astra, 2021-02-06:
Moin moin (greetings to Hamburg from another town at the river Elbe ;-) ),
is it possible to debug the application via st-link v2 in segger embedded studio?

I could compile the code as it is and program it to an STM32F103VET6 eval board. But at first I'd like to check, what the code does and which GPIO are set. Markus, could you tell us, what else you have connected to your board.

The CAN bus control leds are external connected to B13, B14, right?. Is it also possible to use the leds on the eval board. In my case it should be E1 and E0.

[image]

Are there any other hints to start with your excellent reengineered work?

BR maik

sarnau, 2021-02-06:
I am using a SEGGER J-Link PRO, which I found at eBay UK several years ago. So, debugging with that one is a non-issue. You can also flash a st-link v2 to a J-Link, which helps.

davidwallis3101, 2021-02-06:
I think only onboard st link can be flashed, not the usb dongle version??

MS-Astra, 2021-02-08:
Thanks for your feedback. I think the SEGGER J-Link EDU MINI is for me the best option.

I tried the example (with the DI and relay extension) as it is, but the miniserver doesn't detect any extension.
As I understood the legacy extensions doesn't need the AES keys. So this could not the reason that it doesn't work.

Any idea?

davidwallis3101, 2021-02-11:
@MS-Astra  based on your comment, I bought the J-Link EDU mini - and glad I did as previously I was looking at the clones due to the cost.

Can confirm that this works fine, had to fiddle with a couple of options in the project to get it working, but can now debug.. 

Just need to get an CAN bus connection to my desk now and try and get a bit further.

MS-Astra, 2021-02-11:
hey @davidwallis3101,
sounds good. Please, let me know if you can run LoxLink (build with SEGGER) successful with the miniserver on a STM32 board.

davidwallis3101, 2021-02-12:
I'm connected the board to loxlink but not getting too far at the moment, need to try and find a schematic for the board I have which is 

[image]

can then try and find out what LED is what, and that the two jumpers marked 0 and 1 are..

David

MS-Astra, 2021-02-12:
[image]
This is my board. Looks a little bit different and has no jumpers etc.

I got all information of my board from the seller as download link. So the data should be valid.

At the moment I have no idea why it isn't running. Maybe I can use the rs485 to print some debug output as long I have no real debugger...

matlab22, 2021-02-13:
> 
> 
> I'm connected the board to loxlink but not getting too far at the moment, need to try and find a schematic for the board I have which is
> 
> [image]
> 
> can then try and find out what LED is what, and that the two jumpers marked 0 and 1 are..
> 
> David

I started with the same board as you. The jumpers are for the boot mode of the stock boot loader of the stm32. if boot0 is set to 0 it does not matter what settings you have on boot1. when the two leds are blinking, the application is running, so you are not in the boot loader.

davidwallis3101, 2021-02-13:
I've not got as far as blinking leds, any ideas what io is allocated to d2
and d3?

MS-Astra, 2021-02-14:
I came a step further. But it is not enough.

@sarnau Can you simply add two virtual extension, one which uses NAT and one which uses the legacy protocol? These virtual extensions should not connected to any real gpio and could send for instance fixed values, so we can easily test the communication on any STM32 board.

On my board the current configured DI-extension uses the LED-GPIOs... so this can't work.

Or is there an other simple way to test the CAN communication with the miniserver?

matlab22, 2021-02-14:
Even without proper key you should be able to see your extension. Could it be that you are missing the termination resistor?

matlab22, 2021-02-14:
> 
> I've not got as far as blinking leds, any ideas what io is allocated to d2 and d3?
> […](#)

Your boards LED are on GPIOB, GPIO_PIN_13 and GPIO_PIN_14, as in the example from sarnau. When the LEDs aren't blinking the application does not start. You are ether in boot mode or facing the same problem I did: https://github.com/sarnau/LoxLink/issues/2

davidwallis3101, 2021-02-14:
same problem you did, my Prerrocessor definitions now look like this:

```
__CROSSWORKS_ARM
STARTUP_FROM_RESET
STM32F103xE
NESTED_INTERRUPTS
CTL_TASKING
USE_PROCESS_STACK
```

MS-Astra, 2021-02-14:
> @MS-Astra Even without proper key you should be able to see your extension. Could it be that you are missing the termination resistor?

No, the termination resistor is on the board. I thought maybe CAN H and CAN L is mixed up. Or something of the initialization part is already wrong.

[image]

matlab22, 2021-02-14:
@MS-Astra It looks like you have the CAN_RX and TX on diffrent MCU pins.
Try to adjust this in LoxCANDriver_STM32.cpp:
#define CAN_GPIO_PORT GPIOB
#define CAN_RX_GPIO_PIN GPIO_PIN_8
#define CAN_TX_GPIO_PIN GPIO_PIN_9

MS-Astra, 2021-02-14:
> @MS-Astra It looks like you have the CAN_RX and TX on diffrent MCU pins.
> Try to adjust this in LoxCANDriver_STM32.cpp:
> #define CAN_GPIO_PORT GPIOB
> #define CAN_RX_GPIO_PIN GPIO_PIN_8
> #define CAN_TX_GPIO_PIN GPIO_PIN_9

@matlab22 Please take a deeper look at my [changes](https://github.com/MS-Astra/LoxLink/pull/1/files)

MS-Astra, 2021-02-14:
I have the same behaviour that the program doesn't start, so I have to press the reset button and then the LEDs are blinking.

adding STARTUP_FROM_RESET doesn't help

davidwallis3101, 2021-02-15:
I've had a play with this today and pulling my hair out.. have traced the io on the board and the gpio for CAN is correct on PB8 and PB9 

LED's are PB13 and PB14 but I'm getting effectively a flashing red led, IE both flashing and dont see any CAN comms..

Have a gnd connected between mini server and this PCB, CAN HI and Can LOW connected - have tried reversing polarity..

can anyone confirm which is CAN HI on an extension, Left or Right.. did check with the scope earlier but it's my dads scope and due to the current situation I cant get him here to help me work the damn thing properly.. too frigging complicated to get the damn thing to trigger but that's another story!

Bit of a loss as to what to check next other than wonder if the can interface even works on this board!

I do have another board without the onboard can and a seperate MCP2515(iirc).. board I could try I guess..

MS-Astra, 2021-02-16:
@davidwallis3101 can you check if the application could finish the init part for CAN here in [LoxCANDriver_STM32.cpp#L126](https://github.com/sarnau/LoxLink/blob/master/Project/application_code/Loxone/CAN%20Driver/LoxCANDriver_STM32.cpp#L126)

And my next idea is that we should check if all tasks are running. The LED task is running otherwise it will not blink all the time.

davidwallis3101, 2021-02-16:
Yeah it hits the breakpoints when I place them in there..

I added some of the debugging back in too..

[image]

MS-Astra, 2021-02-16:
[image]

the remapping was the cause, look at the [commit](https://github.com/MS-Astra/LoxLink/pull/1/commits/a496c780a45ab315424f03f3d3c41863b66049ee)

`/**
  * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
  * @note  CASE 1: CAN_RX mapped to PA11, CAN_TX mapped to PA12
  * @retval None
  */
#define __HAL_AFIO_REMAP_CAN1_1()  AFIO_REMAP_PARTIAL(AFIO_MAPR_CAN_REMAP_REMAP1, AFIO_MAPR_CAN_REMAP)

/**
  * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
  * @note  CASE 2: CAN_RX mapped to PB8,  CAN_TX mapped to PB9 (not available on 36-pin package)
  * @retval None
  */
#define __HAL_AFIO_REMAP_CAN1_2()  AFIO_REMAP_PARTIAL(AFIO_MAPR_CAN_REMAP_REMAP2, AFIO_MAPR_CAN_REMAP)

/**
  * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
  * @note  CASE 3: CAN_RX mapped to PD0,  CAN_TX mapped to PD1
  * @retval None
  */
#define __HAL_AFIO_REMAP_CAN1_3()  AFIO_REMAP_PARTIAL(AFIO_MAPR_CAN_REMAP_REMAP3, AFIO_MAPR_CAN_REMAP)`

davidwallis3101, 2021-02-17:
glad that works for you,  

With mine the transceiver being on PB8 / 9 I think the original code is correct..

It does look like its failing to send the CAN messages at startup correctly if I'm reading this correctly?

[image]

http://www.disca.upv.es/aperles/arm_cortex_m3/llibre/st/STM32F439xx_User_Manual/stm32f4xx__hal__can_8h_source.html#l00251

David

MS-Astra, 2021-02-21:
Hi David,

on my board I regonized that the communication only works if I use the external power. With the connected ST-Link it doesn't work. Maybe there is a difference.

At the moment the authentification doesn't work in my case. That is my next step.

KjellVerb, 2022-04-11:
Hi,

Due to an SSD crash, I needed to reinstall Segger. Now I face the issue that I get the error `undefined reference to debug_printf` many times and the build will not continue.
Could anyone please go into their project, right-click `debug_printf(...);` in any one of the files and tell me the result of "go to definition"?

Kjell

matlab22, 2022-04-12:
> Hi,
> 
> Due to an SSD crash, I needed to reinstall Segger. Luckily I was able to retrace all my steps based on the document I made and shared above. However, now I face the issue that I get the error `undefined reference to debug_printf` many times and the build will not continue. It may be due to the fact that Crossworks and Segger are much newer versions now. Could anyone please go into their project, right-click `debug_printf(...);` anywhere in the code and tell me the result of "go to definition"? If I find a way to fix this, I'll add it to the file.
> 
> Kjell

[image]
This is part of debugio.h.
I was never able to run it in segger studio. So if you manage to get it build again, do you mind sharing the project files?

KjellVerb, 2022-04-12:
Thanks for checking. I have that file included but in debugio.h it's only declared not defined, so I that's why I get the "undefined" error.
I used to have debug_printf() working in Segger but apparently didn't write down how I got there. I'll let you know if I do.

davidwallis3101, 2022-04-12:
I’ve just had a look at the notes I made a while ago on getting this working on windows.. (I’ve not had chance to touch this in a long time though):some may help point you in the right direction (obv paths will need fixing)Download and install Segger Embedded Studio for ARMDownload and install Crossworks for ARM evaluation (code size limited) version -  https://cdn.rowleydownload.co.uk/arm/releases/arm_crossworks_4_7_3_win_x86_setup.exe Open Crossworks and install following packages via the package manager:                - ARM CPU support package                - CMSIS 5 CMSIS-Core(M) support package                - Crossworks Tasking Library                - STM32F10x Standard Peripherals Library Updates --> C:\Users\Davidw\AppData\Local\SEGGER\SEGGER Embedded Studio\v3\packages\targets\STM32F10x Copy the packages directory from Crossworks to Segger, on Mac these are                C:\Program Files\Rowley Associates Limited\CrossWorks for ARM 4.7                 - ~/Library/SEGGER/SEGGER Embedded Studio/v3/packages                        - ~/Library/Rowley Associates Limited/CrossWorks for ARM/v4/packages                 Xcopy "C:\Users\Davidw\AppData\Local\Rowley Associates Limited\CrossWorks for ARM\v4\packages\*.*" "C:\Users\Davidw\AppData\Local\SEGGER\SEGGER Embedded Studio\v3\packages" /s /e Merge the include directory from Crossworks into Segger, replacing all newer files                - ~/Applications/Segger.../include                on Windows this is in ProgramData (?)  XCopy "C:\Program Files\Rowley Associates Limited\CrossWorks for ARM 4.7\include\*.*" "C:\Program Files\SEGGER\SEGGER Embedded Studio for ARM 5.34\include"  /s /e /y Clone the LoxLink and ctl_fifo projects to your PC git clone https://github.com/sarnau/ctl_fifo.gitCopy ctl_fifo to Loxlink/Project/lib/ Execute downloadLoxoneAESKeys.py Change the extension of LoxLink_CW4.6.hzp to *.emproject and open itren LoxLink_CW4.6.hzp LoxLink_CW4.6.emprojectren LoxLink.hzp LoxLink.emproject Change the Project options for common build configuration                Build                                - Correct memory map file                                - Correct DeviceVectorsFile under Project Macros                                - Correct target processor                Debugger                                - Change whatever is necessary                 Preprocessor definitions in .emprojectfile                - Change "STM32F103x. to the correct one for your board                - add #define __CROSSWORKS_ARM Tools - options - building - Global Macros: add "TargetsDir=/Users/../Library/SEGGER/SEGGER Embedded Studio/v3/packages/targets" or Windows equivalent folder TargetsDir=C:\Users\Davidw\AppData\Local\SEGGER\SEGGER Embedded Studio\v3\packages\targets – works David From: KjellVerbSent: 12 April 2022 07:38To: sarnau/LoxLinkCc: David Wallis; MentionSubject: Re: [sarnau/LoxLink] Segger Embedded Studio for Arm (#9) Thanks for checking. I have that file included but in debugio.h it's only declared not defined, so I that's why I get the "undefined" error.I used to have debug_printf() working in Segger but apparently didn't write down how I got there. I'll let you know if I do.

KjellVerb, 2022-04-12:
I gave up and just added a return 0 function to solve the debug_printf() error. I'll dearly miss the easy debug prints, luckily I did most of my development already in the past when the prints were still working for me.

>I was never able to run it in segger studio. So if you manage to get it build again, do you mind sharing the project files?
I don't think this will help you because you need to get some libraries imported etc. Where did you get stuck?

challo2018, 2022-06-05:
@KjellVerb : I also managed to get it running in Segger Studio. But struggeling with same debug_printf() error. 
Im starting development so would be very good to have it running. Any idea how to fix it?

challo2018, 2022-06-07:
Yes you can get it running with that. But then it is not possible to use debug outputs for debugging. That would be very helpful.

KjellVerb, 2022-06-07:
I agree so if someone is able to fix it, I'm very curious to find the solution!
I once had it working but then my SSD crashed and after reinstalling everything I never got debug working again. Luckily I had done most of my debug/development already, and with my ST-link v2 I can use breakpoints and watches which are good enough for debugging.

IoTSkills, 2022-07-19:
Hello!
Could you recommend a good and cheap debugger/programmer for Segger Embedded Studio? I want to upload this project to ST. Thanks

sarnau, 2022-07-19:
I was lucky to find a Segger J-Link PRO on eBay a couple of years ago, which works great. I think on eBay you can find EDU versions for pretty cheap.

IoTSkills, 2022-07-19:
> I was lucky to find a Segger J-Link PRO on eBay a couple of years ago, which works great. I think on eBay you can find EDU versions for pretty cheap.

Hi! Thanks for the tip, in what environment should this project be used? What do you suggest?

sarnau, 2022-07-19:
CrossStudio or SEGGER both work. I used CrossStudio, because it seems Loxone is using it as well – based on my analysis. SEGGER IDE is just a rebranded version. The licenses are a bit different. I think SEGGER has a free for hobbiest ones. CrossStudio requires a “dongle” (tied to an Ethernet MAC, so I bought a cheap USB Ethernet adapter and tie it to that one, so it is not bound to the actual hardware).

IoTSkills, 2022-07-19:
I followed the steps in the TXT, do you have any solution for this?
[image]

sarnau, 2022-07-19:
That depends on the IDE. SEGGER and CrossStudio doing it differently, you have to check their debugger support (try with an example project, like hello world)

challo2018, 2022-07-19:
i changed it for segger:

#include <SEGGER_RTT.h>
SEGGER_RTT_WriteString(0,"LED blinking orange\n");

[6 more replies]