sarnau/LoxLink #10: Hardware aufbau

feurer98, 2020-02-26:
Hallo Markus

Ich habe mir ein Ziel gesetzt eine Relai & DI Extension aufzubauen. Einerseits als Projektarbeit für die Ausbildung und anderseits aus Interessen, da ich Zuhause einen neuen Server stehen habe.

Nun ist es für mich ein wenig schwer, aus dem Code alle Anschlüsse zu finden und dann auch noch richtig zuzuordnen.
Hast du evtl. eine Pinbelegung des Boards?
Meine beiden bestellten Boards sind noch unterwegs, daher entfällt mir leider noch die Try and Error Methode.

Die Buttons an PB0 und PB15 für was werden diese verwendet?
Die CAN Schnittstelle an PB8 und PB9 verstehe ich richtig, das diese für die LoxLink ist.
Daher müsste eigendlich PA11 und PA12 für die Tree Extension sein. Leider kann ich dies im Code nicht nachvollziehen. 

Bei Interesse kann ich auch den Schaltplan und Layout deinem Projekt beisteuern.
Im Anhang ist vor allem das pdf Haupt_uC interresant, wo ich nach Interpretation des Code die Belegung eingetragen habe.
Das zweite Dokument sollte die Möglichkeit bringen, direkt mit einem USB ohne externen ST-Link das Projekt zu warten. 
[Haupt_uC.pdf](https://github.com/sarnau/LoxLink/files/4255424/Haupt_uC.pdf)
[Programmer.pdf](https://github.com/sarnau/LoxLink/files/4255425/Programmer.pdf)

lg
Benj

matlab22, 2020-05-02:
So wie ich den Code versteh hängt alles an CAN1. Der Controller hat ja nur einen:
#define CAN_GPIO_PORT GPIOB
#define CAN_RX_GPIO_PIN GPIO_PIN_8
#define CAN_TX_GPIO_PIN GPIO_PIN_9
@sarnau Wie ist dies mit der Three extension gedacht? Braucht es hier nicht einen Controller mit zwei Transceiver?

sarnau, 2020-05-02:
Natürlich. Die Tree Extension hat 3: einmal den Loxone Link Bus und zwei Tree Busse.

closed 2020-11-16 by feurer98