sarnau/LoxLink #11: FTP Problem

rem4332, 2020-03-03:
Hallo, ich scheitere am erstellen der AESKeys hat jemand eine IDEE für mich ? 
`ftp.retrbinary("RETR %s_B1E1424BFF667AF471D715EE6745FDF0.upd" % version, callback=handle_binary)
  File "C:\Python27\lib\ftplib.py", line 416, in retrbinary
    conn = self.transfercmd(cmd, rest)
  File "C:\Python27\lib\ftplib.py", line 378, in transfercmd
    return self.ntransfercmd(cmd, rest)[0]
  File "C:\Python27\lib\ftplib.py", line 341, in ntransfercmd
    resp = self.sendcmd(cmd)
  File "C:\Python27\lib\ftplib.py", line 251, in sendcmd
    return self.getresp()
  File "C:\Python27\lib\ftplib.py", line 226, in getresp
    raise error_perm, resp
error_perm: 550 Can't open ._B1E1424BFF667AF471D715EE6745FDF0.upd: No such file or directory
>>> `

Vielen Dank 
lg

sarnau, 2020-03-03:
It seems `version` is empty for you.

rem4332, 2020-03-03:
ja das hab ich gefunden aber wenn ich nun die Version eintrage 
`ftp.retrbinary("RETR %s10031127_B1E1424BFF667AF471D715EE6745FDF0.upd" % version, callback=handle_binary)`
kommt trotzdem das er das File nicht findet. Weil er mir immer eine Punkt rein macht.
Wenn ich es so mache 
`ftp.retrbinary("RETR %10031127_B1E1424BFF667AF471D715EE6745FDF0.upd" % version, callback=handle_binary)`
kommt 
`ftp.retrbinary("RETR %10031127_B1E1424BFF667AF471D715EE6745FDF0.upd" % version, callback=handle_binary)
ValueError: unsupported format character '_' (0x5f) at index 14`

Vielleicht noch eine Idee ? Kann es sein das durch ein Downgrade des neuen MIniservers ich die AES Keys gar nicht erzeugen kann ? Hab zwar wieder auf die 10.3.11.27 upgedated aber man weiß ja nie 

Vielen Dank

sarnau, 2020-03-03:
Du mußt die `version` Variable setzen, nicht die Version in diesen String reinpappen.

sarnau, 2020-03-03:
Den neuen Miniserver habe ich noch nicht gesehen

rem4332, 2020-03-03:
Upps jetzt läufts 
danke manchmal sieht man vor lauter Bäumen den Wald nicht mehr

rem4332, 2020-03-03:
Vielen Dank

closed 2020-03-03 by rem4332