sarnau/LoxLink #13: CrossStudio und ST-Link

feurer98, 2020-03-09:
Hallo Zusammen

Hat wer von euch schon das Projekt mit einem ST-Link Flashen können?

Bei mir sind nun meine [Boards ](https://de.aliexpress.com/item/32911971615.html?spm=a2g0o.productlist.0.0.164d19945MjP4r&algo_pvid=62e328f2-4938-4d48-b632-855df22824ce&algo_expid=62e328f2-4938-4d48-b632-855df22824ce-40&btsid=336657ad-38d4-4cf2-9ad2-ca4ae7ae93ad&ws_ab_test=searchweb0_0,searchweb201602_5,searchweb201603_53)angekommen und ich kann nun auch Testen.
Mit den ST-Link und uVision konnte ich verifizieren, dass ich das Board beschreiben kann und der Treiber auch korrekt funktioniert.
Das File von @matlab22 habe ich via STM32CUBE Programmer auch getestet und den Pin PC13 gemessen.

Beim uploade-Versuch aus CrossStudio bekomme ich jedoch einen DEV_AP_ACCESS_ERROR.
`Loading target script file STM32_Target.js
Executing match part name script
Executing script MatchPartName("STM32F103VE")
Loading target script file STM32_Target.js
Preparing target for download
Executing Reset script Reset()
Erasing ‘LoxLink.elf’ on ST-LINK/V2
Erase successful
Downloading ‘LoxLink.elf’ to ST-LINK/V2
Programming 51.0 KB of addresses 08000000 — 0800cc3f
Programming 0.0 KB of addresses 0800cc40 — 0800cc58
Download failed
`

Beim Manuellen laden der elf Datei, via CubeProgrammer bekomme ich auch einen Fehler: Tow ore more semgents defines the same memory zone.
`\LoxLink-master\Project\LoxLink Debug\LoxLink.elf
10:55:41 : Number of segments: 2
10:55:41 : segment[0]: address= 0x8000000, size= 0xCC59
10:55:41 : segment[1]: address= 0x2000001C, size= 0x0
10:55:41 : Warning: File corrupted. Two or more segments defines the same memory zone`

Ich vermute nun, dass eine Einstellung für den St-Link nicht stimmt. Jedoch verwirren mich der andere Error noch zusätzlich.

lg
Benjamin

challo2018, 2020-03-11:
In Crossstudio bekomme ich auch diesen Fehler. Hab auch mit dem Support versucht hier eine Lösung zu finden. Aber hier gibt es keine Lösung, meldet sich niemand mehr...
Wenn ich das Projekt in Crossstudio compiliere und mit ST Link Utility flashe läuft es bei mir.  Debuggen funktioniert aber dann leider nicht

feurer98, 2020-03-11:
@challo2018

Ich habe es gestern Abend noch hingekriegt.
Ich schicke heute Abend noch die Einstellungen. (Auch wen ich jetzt schon nicht mehr weiss, was ich alles geändert habe).

Arbeitest du auch mit Crossstudio 4.6?
Bei mir hat der support nur darauf hingewiesen den ST-Link auf die neuste Firmware zu kontrollieren.

feurer98, 2020-03-12:
@challo2018 Leider hat es mir gestern nicht mehr geklappt. Ich habe dir mein Projekt angehängt, welches ich mit CrossStudio 4.6 und den ST-Link V2 zum laufen gekriegt habe. 
Ich musste noch eine Zeile auskommentieren, frag mich aber nicht welche :( 

[Github Crosswork46.zip](https://github.com/sarnau/LoxLink/files/4322789/Github.Crosswork46.zip)

closed 2020-03-12 by feurer98