sarnau/LoxLink #14: Tree Device

chrisrock1984, 2020-03-22:
Hi,
Ich habe mal versucht ein Tree Device zu kompilieren. Es ist aber nur möglich, wenn auch die Tree Extension mit kompiliert wird.
Ist es also richtig, dass man kein Tree Device stand-alone kompilieren kann?

sarnau, 2020-03-22:
Ja, den Fall habe ich nie implementiert – ist aber machbar. Ich sehe denn Sinn nicht, denn man kann dem Miniserver ja mehr als eine Tree Extension unterschieben und somit einfach diverse Tree Devices und die Extension auf einem STM32 simulieren.

chrisrock1984, 2020-03-23:
Nur für den unwahrscheinlichen Fall, dass man mehrere Tree Devices baut und die maximale Anzahl an LoxBus devices überschreitet ;-)

challo2018, 2020-03-23:
Ich habe auch versucht nur ein Tree device zu kompilieren. funktioniert bei mir problemlos und es wird auch auf dem tree bus von der tree extension erkannt. allerdings bekomme ich nach einiger zeit das authentifizierungsproblem angezeigt...

sarnau, 2020-03-24:
Was meinst Du mit "einiger Zeit"? Tage? Wochen? Monate? :-)

challo2018, 2020-03-24:
Minuten. Sorry wegen der ungenauen Angabe...

chrisrock1984, 2020-03-24:
Kann ich bestätigen. 
Wenn ich aber nur die Tree Extension einfüge ohne Tree Device funktioniert es. Sobald ich aber das Tree Device suche und hinzufüge bekomme ich viele Fehler sowie den Authentication Error.

challo2018, 2020-03-24:
Ich hab das ganze ohne tree Extension direkt auf dem tree Bus getestet. Wird problemlos erkannt aber dann kommt der authentication Fehler

chrisrock1984, 2020-03-24:
Wie, du betreibst dein STM32 board als Tree Device? Verstehe ich das richtig, dass du eine originale Tree Extension hast und dein Board hängt am Tree Bus?

challo2018, 2020-03-24:
Genau hab das Board auf den Tree Bus hinter einer original Extension gehängt. 
Folgende Anpassungen habe ich gemacht in der main:

static LoxCANDriver_STM32 gLoxCANDriver(**tLoxCANDriverType_TreeBus**);

static LoxBusTreeAlarmSiren gLoxBusTreeAlarmSiren(**gLoxCANDriver**, 0xb010035c, gResetReason);

chrisrock1984, 2020-03-24:
Das probier ich gleich mal aus ;-)

chrisrock1984, 2020-03-24:
CAN H = grün, oder?
Ich hoffe, dass **DER Guru** noch die Authenticate Probleme löst ;-)

challo2018, 2020-03-24:
Kann gerade nicht nachschauen. Ich habs einfach ausprobiert. Sollte nichts passieren wenn man es vertauscht.

Wäre cool wenn du es auch mal testen könntest! Bei der AlarmSirene ist vielleicht noch wichtig das der Sabotage Kontakt als Okay gesendet wird. Ansonsten meldet Loxone Sabotage.
-> hardwareTamperStatusOk

chrisrock1984, 2020-03-24:
Gerade getestet - Auth. Fehler nach ca 1 Min.
Nach dem Neustart sagt die Tree Diagnose auch, dass 100% der Pakete verloren gegangen sind, obwohl das Tree Device ordnungsgemäß erkannt wurde.

challo2018, 2020-03-24:
Super, vielen Dank! Dann hab ich erstmal nichts falsch gemacht. Wäre echt super wenn wir hier den Fehler finden könnten.

chrisrock1984, 2020-03-24:
Könnte man das "issue" wieder auf Open setzen?

sarnau, 2020-03-24:
Die Authorization Issue gibt es noch "offen". Dies ist somit ein Dupe

closed 2020-03-24 by sarnau

challo2018, 2020-03-25:
@sarnau Können wir irgendwie bei der Fehleranalyse unterstützen?

chrisrock1984, 2020-03-25:
Genau, helfe auch gerne mit ;)