sarnau/LoxLink #15: CAN bus TX error - arduino mega - mcp_can

janbalcarik, 2020-05-20:
Good day,

I ported your code to the ArduinoMEGA + MCP_CAN device. Unfortunately I have a problem on the CAN bus and I don't really understand where the problem is. Haven't you seen this problem in your development?

The error always occurs after several sent messages.

Loxone log:

```
00000031 192.168.1.77    22:06:20.808 22:07:01.263 LNK NAT index request from extension 14111779 (device type: 0014)
00000032 192.168.1.77    22:06:20.808 22:07:01.263 LNK send NAT offer to Extension 14111779, new NAT index: 129, park extension
00000033 192.168.1.77    22:06:22.808 22:07:03.294 LNK got version packet from parked extension 14111779, Version: 20031108
00000034 192.168.1.77    22:06:23.332 22:07:03.818 LNK NAT index request from extension 14111775 (device type: 0014)
00000035 192.168.1.77    22:06:23.332 22:07:03.818 LNK send NAT offer to Extension 14111775, new NAT index: 2
00000036 192.168.1.77    22:06:23.356 22:07:03.839 LNK ESR1 = 0x0007f TX err cnt = 128 RX err cnt = 4
00000037 192.168.1.77    22:06:23.356 22:07:03.840 LNK Error-Passive
00000038 192.168.1.77    22:06:23.977 22:07:04.463 LNK NAT index request from extension 14111777 (device type: 0014)
00000039 192.168.1.77    22:06:23.977 22:07:04.463 LNK send NAT offer to Extension 14111777, new NAT index: 130, park extension
00000040 192.168.1.77    22:06:26.009 22:07:06.495 LNK got version packet from parked extension 14111777, Version: 20031108
00000041 192.168.1.77    22:06:26.533 22:07:07.019 LNK NAT index request from extension 14111778 (device type: 0014)
00000042 192.168.1.77    22:06:26.533 22:07:07.019 LNK send NAT offer to Extension 14111778, new NAT index: 131, park extension
00000043 192.168.1.77    22:06:26.554 22:07:07.039 LNK ESR1 = 0x000e7 TX err cnt = 255 RX err cnt = 5
00000044 192.168.1.77    22:06:26.555 22:07:07.040 LNK Bus-Off
00000045 192.168.1.77    22:06:27.058 22:07:07.544 LNK NAT index request from extension 14111776 (device type: 0014)
00000046 192.168.1.77    22:06:27.058 22:07:07.544 LNK send NAT offer to Extension 14111776, new NAT index: 3
00000047 192.168.1.77    22:06:27.570 22:07:08.056 LNK packet from extension (14111776 / DI Extension)
00000048 192.168.1.77    22:06:28.075 22:07:08.561 LNK packet from extension (14111776 / DI Extension)
00000049 192.168.1.77    22:06:28.581 22:07:09.066 LNK packet from extension (14111776 / DI Extension)
00000050 192.168.1.77    22:06:29.093 22:07:09.578 LNK packet from extension (14111776 / DI Extension)
```

Arduino Log:

```
Tx: 336664441 : 269250814, Command: FE, Data: 00 00 14 00 79 17 11 14 
Send status: 0
Rx: 000000000 : 2423255293, Command: FD, Data: FF 81 01 00 79 17 11 14 , ExtensionNat: FF, value32: 14111779
Tx: 336664441 : 270012656, Command: F0, Data: 00 02 14 00 CA 3B CF CB 
Tx: 336664441 : 270012657, Command: F1, Data: 00 84 A6 31 01 00 00 00 
Tx: 336664441 : 270012657, Command: F1, Data: 00 00 9B EF 49 91 79 17 
Tx: 336664441 : 270012657, Command: F1, Data: 00 11 14 02 14 00 00 17 
Tx: 336664437 : 269152510, Command: FE, Data: 00 00 14 00 75 17 11 14 
Tx: 336664439 : 268972286, Command: FE, Data: 00 00 14 00 77 17 11 14 
Rx: 000000000 : 2423255293, Command: FD, Data: FF 82 01 00 77 17 11 14 , ExtensionNat: FF, value32: 14111777
Tx: 336664439 : 270016752, Command: F0, Data: 00 02 14 00 55 70 C8 38 
Tx: 336664439 : 270016753, Command: F1, Data: 00 84 A6 31 01 00 00 00 
Tx: 336664439 : 270016753, Command: F1, Data: 00 00 9B EF 49 91 77 17 
Tx: 336664439 : 270016753, Command: F1, Data: 00 11 14 02 14 00 00 17 
Tx: 336664440 : 269291774, Command: FE, Data: 00 00 14 00 78 17 11 14 
Tx: 336664438 : 269046014, Command: FE, Data: 00 00 14 00 76 17 11 14 
Rx: 000000000 : 2423255293, Command: FD, Data: FF 03 00 00 76 17 11 14 , ExtensionNat: FF, value32: 14111776
Tx: 336664438 : 269496560, Command: F0, Data: 00 02 14 00 D8 17 C5 71 
Tx: 336664438 : 269496561, Command: F1, Data: 00 84 A6 31 01 00 00 00 
Tx: 336664438 : 269496561, Command: F1, Data: 00 00 9B EF 49 91 76 17 
Tx: 336664438 : 269496561, Command: F1, Data: 00 11 14 02 14 00 00 17
 
Send status: 0, Error register value: 10101, TX error count: 131, RX error count: 0

Rx: 000000000 : 2423271665, Command: F1, Data: 00 10 01 0C 00 D0 02 00 , ExtensionNat: 3, value32: 2D000
Rx: 000000000 : 2423271665, Command: F1, Data: 00 00 00 00 00 00 00 00 , ExtensionNat: 3, value32: 0
Rx: 000000000 : 2423271665, Command: F1, Data: 00 00 00 00 00 00 00 00 , ExtensionNat: 3, value32: 0
Tx: 336664437 : 269152510, Command: FE, Data: 00 00 14 00 75 17 11 14
Tx: 336664440 : 269291774, Command: FE, Data: 00 00 14 00 78 17 11 14
 
Send status: 0, Error register value: 10101, TX error count: 129, RX error count: 0

.....
....
..
```

For the detected errors on the arduin side I use this code:

```
	if (CAN0.checkError() == CAN_CTRLERROR) {
		Serial.print("Send status: ");
		Serial.print(sndStat, HEX);

		Serial.print(", Error register value: ");
		byte tempErr = CAN0.getError(); // We are only interested in errors, not warnings.
		Serial.print(tempErr, BIN);

		Serial.print(", TX error count: ");
		tempErr = CAN0.errorCountTX();
		Serial.print(tempErr, DEC);

		Serial.print(", RX error count: ");
		tempErr = CAN0.errorCountRX();
		Serial.print(tempErr, DEC);
		Serial.println();
	}

```

Thank you

sarnau, 2020-05-20:
I have never done ArduinoMega stuff for the CAN bus (too expensive from the hardware point), but is it possible that the authorization fails?

janbalcarik, 2020-05-20:
I wouldn't say that. Probably some HW problem on the CAN bus, buffer, incompatibility ... I don't know. I guess I'd rather rewrite my arduino code for STM...

sarnau, 2020-05-26:
Maybe an interrupt issue?

closed 2020-09-04 by janbalcarik