sarnau/LoxLink #17: Firmware update

KjellVerb, 2020-09-04:
I have adapted the code to update the firmware version when receiving NAT update packages. This works fine but I can't get the CRC verify to work. This causes the Miniserver to keep retrying until it eventually stops communicating with the Extension. If I then reboot the Miniserver and add the Extension again, it works fine (until the Extension is rebooted).
I would like to get the CRC check working, could you help by providing some more documentation about the following statement?

0x80  |  from Device  |  Confirm success of the verify, Page is 0 and _"the actual 32-bit CRC"_ in the optional data area

KjellVerb, 2020-11-19:
I forgot to set byte 1 of the response packet to 0x80, the verify step is working now with a random CRC. 
Would you like me to issue a pull request with update emulation?

closed 2020-11-19 by KjellVerb

chrisrock1984, 2020-12-14:
> I forgot to set byte 1 of the response packet to 0x80, the verify step is working now with a random CRC.
> Would you like me to issue a pull request with update emulation?

Hi,

Is your update bypass code somewhere available?

BR,

KjellVerb, 2021-08-10:
Sorry for the extremely late reply 😅 

```
void LoxNATExtension::update(const eUpdatePackage *updatePackage) {
  //if (this->state == eDeviceState_parked)
  //  return;
  if (updatePackage->device_type != this->device_type)
    return;
  if (updatePackage->version < this->version)
    return;

  switch (updatePackage->updatePackageType) {
  case eUpdatePackageType_write_flash:
    //debug_printf("\nwrite flash\n");
    break;
  case eUpdatePackageType_receive_crc:
    //debug_printf("receive CRC\n");
    break;
  case eUpdatePackageType_verify:
    //debug_printf("\nverify\n");
    //reply with verify OK and random CRC
    uint8_t replyData[16]; // 16 bytes
    memcpy(replyData , updatePackage , 16);
    replyData[0]=16;
    replyData[1]=eUpdatePackageType_reply_verify_ok;
    send_fragmented_message(Update_Reply,replyData, 16);
    break;

  case eUpdatePackageType_verify_and_reset:
    //debug_printf("\nverify and reset\n");
    SetState(eDeviceState_offline);
    this->version = updatePackage->version;
    break;
  default:
    break;
  }
}
```

You'll see that the version variable is only updated in RAM, I have not found a way to make this persistent. This means that if the firmware version is out of date, the update process will happen each time the device is reset or power cycled.

sarnau, 2022-07-20:
Yes, if might make sense to add some code to store the latest version in an EEPROM/FLASH on the device to avoid this issue.

IoTSkills, 2022-07-20:
> Igen, ha érdemes lenne hozzáadni egy kódot a legújabb verzió EEPROM/FLASH-ban való tárolására az eszközön, hogy elkerüljük ezt a problémát.

Yes, it's a shame it's not in STM. I'm testing it now with a demo board, because unfortunately I can't find a source for pro purchase. Do you happen to know a place where STM32F103VE is in stock? I don't trust AliExpress so much in this area :)

IoTSkills, 2022-07-20:
@KjellVerb I pasted the code, but it still doesn't send the new firmware version, only the old one. I removed const from the definition of version because it is read only and the code does not compile. But he doesn't want to go after that either. Could you share with me a project in which this works? :) Thanks

KjellVerb, 2022-07-25:
I don't remember exactly why, but I changed `#define MAX_FRAGMENT_SIZE` in LoxNATExtension.hpp to `72`. Must mean I needed that change.