sarnau/LoxLink #18: Di-Extension refresh rate

matlab22, 2020-11-06:
I made a custom Di Do extension with galvanic insulation. For the input I am using Si838x chips. 
[image]
I configuration loxone to switch an output, when it detects an input.
[image]
When I shortly press the button the yellow led turns on, but the mcu does not detect it. And therefore does not turn on the output. I have to press the button quit long to properly detect an input. This makes it impossible to detect a double click.

As far as I understand the code. With the help of TIM3 it should update the state of the input pins every millisecond. And send it immediately if there is a change of state.

The "Minimum time interval" settings from the loxone config apps are ignored. Or did I miss something in the code?
[image]

Is there anything I can do to get a faster refresh rate of the input pins? The goal is to detect at least a triple click on my light switch.

matlab22, 2020-11-11:
Could it be that the TIM3 is not correctly configured?
The clocks are configured like this:
```
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.HSEPredivValue = RCC_HSE_PREDIV_DIV1;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL9;

  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
```
As far as I understand it, this leeds to the following configuration:
[image]

The timer then is configured as:
```
g1000HzTimer.Instance = TIM3;
g1000HzTimer.Init.Prescaler = HAL_RCC_GetPCLK1Freq() / 2000 - 1;
g1000HzTimer.Init.Period = 2 - 1; // 2000HZ / 2 = 1000Hz
```
HAL_RCC_GetPCLK1Freq() is now 36MHz and the timer at 72MHz shouldn't the prescaler therefore be configured like this:
`
g1000HzTimer.Init.Prescaler = HAL_RCC_GetPCLK1Freq()*2 / 1000 - 1;`
to get 1kHz clock?

Or did I made a mistake?

KjellVerb, 2020-11-12:
I have done an empirical experiment where the DI extension's (ctl eMainEvents_10ms based) `Timer10ms` function toggles one LED when `lastFrequencyTime >= 1000` and the (TIM3 based) `HAL_TIM_PeriodElapsedCallback` function toggles a different LED when `sMillisecondsCounter >= 1000`.

In this test, the ctl based LED state changes once per second and the TIM3 based LED state changes twice per second. Therefore I conclude that the g1000HzTimer frequency is actually 2000Hz.

How long do you need to press/release the button for a state change to be detected? I did see that the input state is read every 0.5ms (TIM3), but it is sent at a maximum rate of 50Hz (see last lines of LoxBusDIExtension.cpp). This should still be plenty to detect double/triple clicks though. If you have an oscilloscope, you may want to check out the input signal toward the STM pins, maybe there's a real HW issue?
Can you share your schematic or at least the input part of it? I could have a look.

matlab22, 2020-11-12:
Thanks for the test. Unfortunately I do not have an oscilloscope at home. I will check it next time at work.
Here is the current schematic:
[Schematic Prints.PDF](https://github.com/sarnau/LoxLink/files/5529479/Schematic.Prints.PDF)
Mounted is a Si8380PS-IU with a 100ms Low-Pass Debounce Filter. The 2.2nF capacitors are not present.
The time is somehow random. Sometimes it is working almost instantly but most of the time it is around one second.

KjellVerb, 2020-11-12:
Did I read the schematic badly or is +U_DIN not connected to anything? I'm assuming that you are testing the inputs with the on-board switches.

KjellVerb, 2020-11-12:
I was looking in the wrong place because I overlooked the following sentence:

> "When I shortly press the button the yellow led turns on"

Given that the LED is working as expected at the secondary side of the isolator, the isolation circuit seems to work fine. You could check the DC levels during button press to make 100% sure that your input levels are OK, in that case I think it would be safe to say that your issue is indeed in the FW and not HW.

matlab22, 2020-11-12:
Somehow I managed o upload an old version of the schematic. I updated it now and will check the levels. But I think it is a poll problem. When I press the button in burst every 0.5s the yellow led always flashes and sometimes the Di extension recognizes the input and turns on the output of the relay extension.

KjellVerb, 2020-11-12:
I did see that the throttling is actually down to 20Hz instead of 50Hz:
`  if ((this->lastBitmaskSendTime % 50) == 0)`  should be 20
You could increase the update throttling even higher by removing this if-condition altogether, but I still don't think it would explain the behavior that you are seeing.

Something else that caught my eye is that `config.frequencyInputsBitmask` doesn't seem to be initialized anywhere. Frequency updates are sent at a rate of 1Hz, so maybe what you are seeing is related to the DI treating inputs as frequency counters? Quite easy to test by initializing the mask to 0 in `LoxBusDIExtension::Startup(void)`

matlab22, 2020-11-16:
Ok I measured the TIM3 by toggling a PIN whenever HAL_TIM_PeriodElapsedCallback is called. This leads to a 500us period and the correct 2kHz.
But the method LoxBusDIExtension::Timer10ms(void) is called every 50ms to 60ms and not every 10ms. Unfortunately I can not find where this method is getting its clock.
I removed the whole throttle statement and onl left this:
```
    if (this->lastBitmaskSend != this->hardwareBitmask) {
      this->lastBitmaskSend = this->hardwareBitmask;
      this->lastBitmaskSendTime = 0;
      send_digital_value(0, this->lastBitmaskSend);
    }
```
Know it is working and also double and triple clicks are working. Leaving the debouncing stuff to the Si8380PS-IU. But it would be nice to solve the problem with the Timer10ms() not really taking 10ms.

In the clock settings HSI is enabled, I don't see why since the clock is coming from HSE. I disabled it and do not see any difference.

KjellVerb, 2020-11-16:
Did you verify the Timer10ms() period the same way? You say it's called every 50-60ms, is that variable or constant?
The Timer10ms comes from a ctl event that is triggered in Timer_Callback_1000Hz(), which comes from the Systick which comes from the core clock.
If you look at my first reply, you'll see that on my test setup (DI only on a so called bluepill module), I get a nice 1Hz from counting 100 Timer10ms events. Maybe for some reason your MCU is bogged down with other tasks and it gets to the Timer10ms routine too late? I'd be surprised but it's not impossible... 
You could try to build DI only and leave DO out for the moment? If you use Loxone live view you can still debug whether double/triple clicks are correctly detected.

matlab22, 2020-11-16:
Yes I did it the same way:
I aded to `LoxBusDIExtension::Startup(void)`
```
  __HAL_RCC_GPIOD_CLK_ENABLE();
    GPIO_InitTypeDef GPIO_Init;
    GPIO_Init.Pin = GPIO_PIN_7;
    GPIO_Init.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_Init.Pull = GPIO_PULLDOWN;
    GPIO_Init.Speed = GPIO_SPEED_FREQ_LOW;
    HAL_GPIO_Init(GPIOD, &GPIO_Init);
```

and `HAL_GPIO_TogglePin(GPIOD,GPIO_PIN_7);` to the end of void `LoxBusDIExtension::Timer10ms(void)`

It is now a DI extension only with the follow irregular behavior:
[image]

KjellVerb, 2020-11-16:
That's not normal I think. 
I don't have an oscilloscope at home and am not going to the office soon (guess why...) so I can't check my board with the same code. I am only able to test by toggling an LED for now. What I did is change `if (this->lastFrequencyTime >= 1000)` in Timer10ms to 60000 and toggle the LED inside the if. I timed the on-time of the LED and it's exactly 60s.
I'm afraid that since I can't reproduce the issue, I can't help you debug.

matlab22, 2020-11-18:
Its not working for me. I tried the same code on a STM32 Board from [Aliexpress](https://www.aliexpress.com/item/32693281117.html?spm=a2g0s.9042311.0.0.27424c4dAOwNAZ)

To get a propper 1kHz from Timer 3, the period should be 4 not 2:
```
  g1000HzTimer.Init.Prescaler = HAL_RCC_GetPCLK1Freq() /2000 - 1;
  g1000HzTimer.Init.Period = 4 - 1; // 2000HZ / 2 = 1000Hz
```

Then I went to confirm the frequency of the Timer10ms. Therefore I modified the code as follow:

```
/***
 *  10ms Timer to be called 100x per second
 ***/
void LoxBusDIExtension::Timer10ms(void) {
  LoxNATExtension::Timer10ms();
  HAL_GPIO_TogglePin(GPIOD,GPIO_PIN_6);
  if (this->lastFrequencyTime >= 100) { // frequencies are sent once per second
    this->lastFrequencyTime = 0;
    HAL_GPIO_TogglePin(GPIOD,GPIO_PIN_7);
  }
  this->lastFrequencyTime += 10;
}
```
[image]
instead of 10s I got 7.7s , so it is somehow accurate. But not what I would expect from an RTOS

It looks like it is of by 30 ticks:
[image]

The SysTick should be 1ms, in fact it is around 10ms and not very accurate:
```
/***
 *  SysTick is called with 1000Hz (every 1ms)
 *
 *  From that we trigger two events:
 *  - every 10ms for short delays
 *  - every second for long delays, like timeouts
 ***/
static void Timer_Callback_1000Hz(void) {
  ctl_increment_tick_from_isr();
  //HAL_IncTick(); // this should not be necessary, because we do not need HAL functions, which rely on this
  static int msCounter = 0;
  if((msCounter % 10) == 0) {
    ctl_events_set_clear(&gMainEvent, eMainEvents_10ms, 0x00); // 10ms event
    HAL_GPIO_TogglePin(GPIOD,GPIO_PIN_6);
  }
  ++msCounter;
  HAL_GPIO_TogglePin(GPIOD,GPIO_PIN_7);
}
```
[image]
I think the problem is with the definition of this clock:
```
#ifdef USEPROCESSORCLOCK
  SysTick->CTRL = SysTick_CTRL_CLKSOURCE_Msk | SysTick_CTRL_TICKINT_Msk | SysTick_CTRL_ENABLE_Msk;
#else
  SysTick->CTRL = SysTick_CTRL_TICKINT_Msk | SysTick_CTRL_ENABLE_Msk;
#endif

In my project USEPROCESSORCLOCK is not defined.

KjellVerb, 2020-11-18:
> To get a propper 1kHz from Timer 3, the period should be 4 not 2:

Agreed. I don't think it's required for correct operation of the DI, but it's nice when reality matches code comments.

>The SysTick should be 1ms, in fact it is around 10ms and not very accurate:

I have also noticed some time ago that SysTick-based HAL delay functions are off by a factor 10, but somehow the ctl delays are correct. I searched for a while but didn't find a reason and eventually let it go.

One thing that I noticed in your measurements: all of the measurements with default config have a variable period, even the one that has once a period of 100ms. Only the third image shows a nice and consistent clock period even if it's not the expected frequency. Did you check the DI reaction speed with `SysTick->CTRL = SysTick_CTRL_CLKSOURCE_Msk | SysTick_CTRL_TICKINT_Msk | SysTick_CTRL_ENABLE_Msk;`?

On my setup USEPROCESSORCLOCK is not defined, but the behavior does not change when I do define it: still get a clock period that is perfectly in line with expectations (as far as I can measure it.. now using a UART message to my PC and comparing timestamps, resolution is not great but it's the best I can do without logic analyzer or scope). 

I'm starting to think that your sysclk config may be wrong. What is printed at the beginning in debug mode? I get this:
`SysClock:72MHz HCLK:72MHz PCLK1:36MHz PCLK2:72MHz`

matlab22, 2020-11-18:
Where do I get this info? SysClock:72MHz HCLK:72MHz PCLK1:36MHz PCLK2:72MHz

KjellVerb, 2020-11-18:
Seems correct, I'm out of ideas for now I'm afraid...

matlab22, 2020-11-18:
STM32_ctl.c
Has a very strange entry:
```
extern uint32_t SystemCoreClock;
#if defined (STM32W108C8) || defined (STM32W108CB) || defined (STM32W108CC) || defined (STM32W108CZ) || defined (STM32W108HB)
#define USEPROCESSORCLOCK
#endif

#ifdef USEPROCESSORCLOCK
#define SYSTICKDIVIDER 1
#else
#define SYSTICKDIVIDER 8
#endif

#define ONE_MS (SystemCoreClock/SYSTICKDIVIDER/1000)
#define TEN_MS (ONE_MS*10)
```
SystemCoreClock=72000000
SYSTICKDIVIDER=8

So one MS should be 9000Hz

So when I correct SYSTICKDIVIDER to 72, I around 10ms ticks but the are not as accurate as the one from TIM3

KjellVerb, 2020-11-19:
You are using the correct values but not the correct calculation. ONE_MS is not a value in Hz but it's the number of ticks of SysTick_CLKSource needed to count one millisecond.
SysTick_CLKSource is HCLK/8 by default, so the reference is 72Mhz/8 = 9MHz, you need 9000 ticks for one millisecond.
If you set USEPROCESSORCLOCK, the reference is 72Mhz, you need 72000 ticks for one millisecond.
So all of the above seems correct to me.

matlab22, 2020-11-19:
hmm could you do me a favor and compile your solution with my GPIOs in the attached file
[system.zip](https://github.com/sarnau/LoxLink/files/5565704/system.zip) and send me the hex
I would like to check if your setup is doing the same -> Timer_Callback_1000Hz is 10ms and ctl_events_set_clear is called every 100ms

matlab22, 2020-11-19:
I guess I found the problem.
In STM32_ctl.c SysTick is actually loaded with 10ms and not 1ms, therefore this timer is not 1kHz. This is making everything 10x slower and therfore my doublecklicks are missing.

```
void ctl_start_timer(CTL_ISR_FN_t timerFn)
{    
  userTimerISR = timerFn;    
  SysTick->LOAD = TEN_MS-1;
  SysTick->VAL = 0;
```

KjellVerb, 2020-11-19:
OMG I'm so sorry I did not see this before, I have already adapted this in the past but it doesn't show up in my repo because it's an externally included file. It's been so long ago that I didn't even remember changing it!

matlab22, 2020-11-19:
👯 Finally 👍 
I think this will happen again. Is there no other way to set this to 1ms then editing the file?

KjellVerb, 2020-11-19:
You could keep STM32_ctl.c as it is and change `Timer_Callback_1000Hz` to always trigger the 10ms event. Preferably also change the function name to `Timer_Callback_100Hz` and change the `ctl_start_timer(Timer_Callback_100(0)Hz)` call accordingly. This way all changes are limited to system.cpp

closed 2020-12-28 by matlab22