sarnau/LoxLink #22: Serial Nummer der Extension eintragen, LEDs blinken nicht...

fl4240, 2021-08-06:
Hallo, ich versuche schon seit längern das ich eine DI extension mit dem STM32F103VET6 herstelle... Leider scheitere ich dabei.

Ich habe die AES Keys ausgelesen und mittels Cross Studio das Programm aufgespielt.

Leider weis ich nicht ob und wo ich die Serien Nummer der DI extension eintragen soll oder muss. 

Die LEDs blinken leider auch nicht. Ich habe bei den preprocessor properties bereits das STARTUP_FROM_RESET eingetragen jedoch hilft das nichts.

Da ich ein ziemlicher Anfänger bin weiß ich leider nicht mehr weiter. 

Vielleicht kann mir jemand ein paar Tipps geben.

Danke im Voraus Franz

fl4240, 2021-08-10:
Hallo, mitlerweile habe habe ich den Code zum laufen gebracht. Zumindest blinken die LEDs. Das Problem ist das Code eine Pause macht, bei b500 push {lr}  siehe Bild im Anhang. Wie kann ich das ändern sodass der Code ohne Pause läuft? Danke Franz 
 [https://drive.google.com/file/d/1HClcJEbdM-HAa_tAwDUa6lQNn97J78Cs/view?usp=sharing](url)

KjellVerb, 2021-08-10:
My guess is that you are building with the "debug" config instead of the "release" config. Anyway when you don't connect the debug interface and reset the STM, it should run without pausing.

fl4240, 2021-08-11:
Hello thank you for the information. I have changed from debug to Release, now it runs without pausing and the Miniserver detects the DI Extension. Perfect!

closed 2021-08-11 by fl4240