sarnau/LoxLink #23: GPIO´s anders zuordnen

fl4240, 2021-09-22:
Hallo, ist es möglich die GPIO´s anders zuzuordnen? 
Ich habe eine fertige SPS Platine ( FX3U-56MR) mit dem STM32F103VCT6 diesen habe ich bereits soweit das ich den Code aufspielen kann. Leider stimmen die GPIO´s der Relais und der Eingänge nicht mit denen des Codes überein. Somit wäre es interessant ob bzw. wo und wie ich die GPIO´s anders zuordnen kann. Der CAN müsste zum Beispiel auf PD0  und PD1 das Run LED auf PD10. Ich habe schon einiges versucht leide blinkt weder das LED noch kann ich die Platine an den Miniserver verbinden. Danke im Voraus Franz.

KjellVerb, 2021-09-22:
- For the inputs and outputs, change the following in LoxLink/Project/application_code/Loxone/NAT/LoxBusDIExtension.cpp and LoxLink/Project/application_code/Loxone/Legacy/LoxLegacyRelayExtension.cpp to match your board.

```
gDIPins[DI_EXTENSION_INPUTS] = {
  {GPIO_PIN_0, GPIOE},
  {GPIO_PIN_1, GPIOE},
...
```

- For the LED: you mention only one IO, do you only have one LED instead of a red and green one?

- For the CAN, you chose the IOs on a remapped alternate function (datasheet page 34) so you need to use the following function to remap it:
```
 /**
   * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
   * @note  CASE 3: CAN_RX mapped to PD0,  CAN_TX mapped to PD1
   * @retval None
   */
 #define __HAL_AFIO_REMAP_CAN1_3()  AFIO_REMAP_PARTIAL(AFIO_MAPR_CAN_REMAP_REMAP3, AFIO_MAPR_CAN_REMAP) 
 ```
 Open up LoxLink/Project/application_code/Loxone/CAN Driver/LoxCANDriver_STM32.cpp and change the following lines to match your board:
 ```
 #define CAN_GPIO_PORT GPIOB
 #define CAN_RX_GPIO_PIN GPIO_PIN_8
 #define CAN_TX_GPIO_PIN GPIO_PIN_9
 ```
 Then scroll down a bit and add `__HAL_AFIO_REMAP_CAN1_3();` to the `HAL_CAN_MspInit` function. I added it after `HAL_GPIO_Init(CAN_GPIO_PORT, &gpioTX);`.

I hope this all works for you :-)

fl4240, 2021-09-23:
Hello KjellVerb, thank you for your answer, do i have to replace the :       `__HAL_AFIO_REMAP_CAN1_2();`
 with `__HAL_AFIO_REMAP_CAN1_3();` ? And in which file do i have to insert the code for remaping the CAN ? Thanks best regards Franz

KjellVerb, 2021-09-23:
Hi, you indeed have to change *CAN1_2 with *CAN1_3 (or at least I think so, I use a different chip so can't test this for you).

All of the CAN hardware related code is in  LoxLink/Project/application_code/Loxone/CAN Driver/LoxCANDriver_STM32.cpp

fl4240, 2021-09-23:
Hi, i´ll test this today and let you know tomorrow. Thank you.

KjellVerb, 2021-09-27:
Hi, any progress on this?

fl4240, 2021-09-28:
Hello, yes there is. I tested yesterday. But i have to measure the gpios again from the board. Some are wrong. And when I assign the DI extension in Loxconfig, the code no longer runs. I think that some gpios will be used 2x. I still have to go through the code. Thanks Franz

fl4240, 2021-10-06:
Hello, last week i was ill so i testet the code yesterday. The code doesn´t stop when i assign the DI Extension in the Loxconfig, only the Run Led stops. I can switch only 9 relais. Here is the code from the LoxLegacyRelayExtension.cpp 

`#include "LoxLegacyRelayExtension.hpp"
#include "stm32f1xx_hal_gpio.h"
#include "stm32f1xx_hal_rcc.h"
#include "system.hpp"
#include <stdlib.h>

static const struct {
  uint16_t pin;
  GPIO_TypeDef *gpio;
} gRelayPins[RELAY_EXTENSION_OUTPUTS] = {
  {GPIO_PIN_3, GPIOE},
  {GPIO_PIN_1, GPIOE},
  {GPIO_PIN_0, GPIOE},
  {GPIO_PIN_2, GPIOA},
  {GPIO_PIN_6, GPIOA},
  {GPIO_PIN_7, GPIOA},
  {GPIO_PIN_15, GPIOB},
  {GPIO_PIN_12, GPIOD},
  {GPIO_PIN_8, GPIOC},
  {GPIO_PIN_9, GPIOC},
  {GPIO_PIN_8, GPIOA},
  {GPIO_PIN_3, GPIOB},
  {GPIO_PIN_8, GPIOB},
  {GPIO_PIN_9, GPIOB},
};

/***
 *  Update the relays
 ***/
void LoxLegacyRelayExtension::update_relays(uint16_t bitmask) {
  if (this->temperatureOverheatingFlag)
    bitmask = 0;
  this->harewareDigitalOutBitmask = bitmask;
  //  debug_printf("### Relay Status 0x%x\n", this->harewareDigitalOutBitmask);
  for (int i = 0; i < RELAY_EXTENSION_OUTPUTS; ++i) {
    HAL_GPIO_WritePin(gRelayPins[i].gpio, gRelayPins[i].pin, (bitmask & (1 << i)) ? GPIO_PIN_SET : GPIO_PIN_RESET);
  }
}

/***
 *  Constructor
 ***/
LoxLegacyRelayExtension::LoxLegacyRelayExtension(LoxCANBaseDriver &driver, uint32_t serial)
  : LoxLegacyExtension(driver, (serial & 0xFFFFFF) | (eDeviceType_t_RelayExtension << 24), eDeviceType_t_RelayExtension, 2, 10031108), harewareDigitalOutBitmask(0), temperatureForceSend(false), temperatureOverheatingFlag(false), temperatureMsTimer(0), temperature(0) {
}

/***
 *  Setup GPIOs
 ***/
void LoxLegacyRelayExtension::Startup(void) {
  // Configure all outputs
  __HAL_RCC_GPIOE_CLK_ENABLE();
  __HAL_RCC_GPIOA_CLK_ENABLE();
  __HAL_RCC_GPIOB_CLK_ENABLE();
  __HAL_RCC_GPIOD_CLK_ENABLE();
  __HAL_RCC_GPIOC_CLK_ENABLE();
  for (int i = 0; i < RELAY_EXTENSION_OUTPUTS; ++i) {
    GPIO_InitTypeDef GPIO_Init;
    GPIO_Init.Pin = gRelayPins[i].pin;
    GPIO_Init.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_Init.Pull = GPIO_PULLDOWN;
    GPIO_Init.Speed = GPIO_SPEED_FREQ_LOW;
    HAL_GPIO_Init(gRelayPins[i].gpio, &GPIO_Init);`

 I don´t understand why i can´t switch the 5 relais on PA2,  PA6,  PA7,  PA8,  PB3.

The DI extension i haven´t testet yet.

Thanks Franz

KjellVerb, 2021-10-06:
Hi,

PB3 has a dedicated function JTDO so you'll need to do some googling in order to find out how to release it for GPIO use. Here's a starting point: [STM32F103 - PB3 just doesn't work as GPIO](https://community.st.com/s/question/0D50X00009XkZMmSAN/stm32f103-pb3-just-doesnt-work-as-gpio)

For PA* I don't have an idea, these should work according to the datasheet. 
Go through the code and make sure that none of these pins are defined anywhere else. It's probably best to not instantiate other extensions yet to avoid double use of GPIOs.