sarnau/LoxLink #25: Device "clocking frequency" and interrupts

schafran, 2022-09-03:
Thank you very much for this in depth analysis and work (also in the other repository about the miniserver).
If I read this correctly, all the main tasks for devices are done inside the Timer10ms method. Does this mean that the devices essentially run at 100Hz update intervals?

How would you implement functionality for the touch device, I assume you wouldn't just scan the touch state at 100Hz but listen for interrupts of the touch driver. On which layer of this hierarchical structure (with CAN Driver, Extension class, Device class) would you implement the interrupt call?

KjellVerb, 2022-10-13:
Why would you not want to poll the IO state? Take a look at the DIExtension for reference, there is already an implementation there for inputs.

schafran, 2022-10-13:
Of course I would want to poll the IO state. I just had a look at the DIExtension. From what I understand, it polls the GPIO pins at 1000 Hz. Inside the `Timer10ms` it then checks every 5th time (essentially making it a `Timer50ms` running at 20 Hz unlike the comment in the code) whether the GPIO states are different from the last reported state and sends an updated state to the bus if that's the case.

This would mean worst case I have a 50 ms delay from a GPIO pin changing to the DIExtension reporting it to the bus. I guess this is acceptable for most applications?
For touch functionality I have a CapSense SoC with an interrupt pin. I was just wondering how I would implement an update being sent on that interrupt instead of polling it for changed state every 10 ms. Essentially making it push instead of 100 Hz poll. But I guess also here, 10 ms delay would be acceptable

KjellVerb, 2022-10-13:
I'm no software engineer but what I would do in that case is monitor the interrupt pin with a 1000Hz loop and use that as a trigger to read out the input states of the SoC.
My double-tap timeout with Loxone Tree Touch buttons is set to 350ms and I wouldn't want it any faster, so I guess a resolution of 10ms would be plenty.

You can always find out online how to use pin-based interrupts but I'm not sure whether that effort is worth it.

sarnau, 2022-10-13:
Do not forget that measuring at a 1ms accuracy is not equal to the Miniserver being able to deal with it. The communication between an extension and the Miniserver also takes time. I would guess that anything below 50ms is pretty pointless. A faster timer is needed, e.g. for PWM dimming, but that is handled within the extension and does not rely on a communication with the Miniserver.

schafran, 2022-10-13:
Thanks for that insight! I guess I will just proceed with the 10 ms resolution since it means minimal alterations to the code structure. I can always do more tweaking when my DIY system is in the testing phase 😃 

@sarnau, yes of course, I was just referring to one small part in the timeline. The discussion still helped me understand the code architecture better

closed 2022-10-13 by schafran